Xilinx/Vitis-AI #634: Quantization of SlowFast_8x8_R50 (PySlowFast) with Vitis AI Quantizer

MarvS95, 2022-01-14:
Hello,

in my master thesis I'm working on the deployment of a SlowFast-Network (PySlowFast from FAIR) on a Xilinx FPGA for human action recognition. The SlowFast model is given as a python skript, .yaml and .pkl extension. The planned procedure now is to quantize the SlowFast-Model (FP32 --> INT8) and compile the quantized model - both using Vitis AI.

Now I'm searching for solutions for the conversion into the file format needed by the Vitis AI quantizer:

Tf1: frozen_graph.pb
Tf2: .h5 or saved model format
PyTorch: model.pth & model.py

I hope I was able to describe my issue properly. Does anybody know how to convert the SlowFast-Network in a way that I can quantize it using the Vitis AI quantizer?

Thank you very much!

wangxd-xlnx, 2022-01-19:
Hi @MarvS95 
It’s no problem to switch to pth or not. If the model format that can be read and load by Pytorch correctly, it can be quantized by Vitis AI quantizer. In fact, pkl and pth just the suffix is different, there is no difference after loaded by Pytorch.

MarvS95, 2022-01-19:
Thanks for your reply @wangxd-xlnx

PyTorch can read and load the pkl and the model runs correctly on my gpu. So you think I don't need a conversion into pth and Vitis AI can read the pkl?

closed 2022-02-03 by MarvS95