Xilinx/Vitis-AI #636: how i deploy the xmodel on cpu？

white0dew, 2022-01-18:
when i get a xmodel, there is a OP which should be running on CPU.
but I dont know how to deploy a OP on CPU, is there a example for us to follow?
Thanks!

bryanloz-xilinx, 2022-01-20:
https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/plugin-samples/samples/plugin-smfc-cpu

This might help

white0dew, 2022-01-20:
> https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/plugin-samples/samples/plugin-smfc-cpu
> 
> This might help

Thanks for your quickly reply, I mentioned that  Vitis Ai  2.0 is available ， so is there a new pdf for reading？

qianglin-xlnx, 2022-02-16:
Hi @white0dew 
Yes. 
For Vitis AI 2.0 user guide, you can refer to : https://docs.xilinx.com/r/en-US/ug1414-vitis-ai 
For pdf version, refer to : http://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/2_0/ug1414-vitis-ai.pdf

BTW, you can also refer to https://github.com/Xilinx/Vitis-AI/tree/master/demo/Custom_OP_Demo to deploy the OP on cpu.

closed 2022-03-01 by qianglin-xlnx