Xilinx/Vitis-AI #638: TensorFlow 1.15 - Error while building the quantized graph for quantization finetuning of SSD MobileNet v1

samagalhaes, 2022-01-19:
Hi,

I am trying to quantise finetuning an SSD MobileNet v1 using TensorFlow 1.15. I am using Vitis AI 1.4 GPU and after activating the TensorFlow environment, I installed the Object Detection API. For this case, I am using the legacy version train.py of OD API, an changed the trainer.py accordingly:

```Python
    model_fn = functools.partial(_create_losses,
                                 create_model_fn=create_model_fn,
                                 train_config=train_config)
    clones = model_deploy.create_clones(deploy_config, model_fn, [input_queue])
    first_clone_scope = clones[0].scope

    if graph_hook_fn:
      with tf.device(deploy_config.variables_device()):
        graph_hook_fn()
    
    print("\n\nCreating quantizer")
    # *Set the quantize configurations
    from tensorflow.contrib import decent_q
    q_config = decent_q.QuantizeConfig(input_nodes=['FeatureExtractor/MobilenetV1/MobilenetV1/Conv2d_0/BatchNorm/batchnorm/mul_1'],
                                      output_nodes=['concat', 'concat_1'], 
                                      input_shapes=[[-1, 300, 300, 3]])
    # *Call Vai_q_tensorflow api to create the quantize training graph
    decent_q.CreateQuantizeTrainingGraph(config=q_config)
    print("Quantizer created\n\n")

    # Gather update_ops from the first clone. These contain, for example,
    # the updates for the batch_norm variables created by model_fn.
    update_ops = tf.get_collection(tf.GraphKeys.UPDATE_OPS, first_clone_scope)
```

Using this configuration, I should only be quantising the layers between preprocessing and after postprocessing.
However, while building the quantisation graph, the system fails with the following error:

```Bash
Creating quantiser
INFO: Creating the quantise train graph...
2022-01-19 17:52:56.084027: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: SSE4.1 SSE4.2 AVX AVX2 AVX512F FMA
2022-01-19 17:52:56.109313: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 3299990000 Hz
2022-01-19 17:52:56.110559: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55c5758be3f0 initialized for platform Host (this does not guarantee that XLA will be used). Devices:
2022-01-19 17:52:56.110591: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Host, Default Version
2022-01-19 17:52:56.114631: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcuda.so.1
2022-01-19 17:52:56.355573: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55c5758c3500 initialized for platform CUDA (this does not guarantee that XLA will be used). Devices:
2022-01-19 17:52:56.355609: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Quadro P2000, Compute Capability 6.1
2022-01-19 17:52:56.356418: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1639] Found device 0 with properties: 
name: Quadro P2000 major: 6 minor: 1 memoryClockRate(GHz): 1.4805
pciBusID: 0000:65:00.0
2022-01-19 17:52:56.363857: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
2022-01-19 17:52:56.583389: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcublas.so.10.0
2022-01-19 17:52:56.699366: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcufft.so.10.0
2022-01-19 17:52:56.741527: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcurand.so.10.0
2022-01-19 17:52:56.983531: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusolver.so.10.0
2022-01-19 17:52:57.132016: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcusparse.so.10.0
2022-01-19 17:52:57.457180: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudnn.so.7
2022-01-19 17:52:57.458925: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1767] Adding visible gpu devices: 0
2022-01-19 17:52:57.459043: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
2022-01-19 17:52:57.479351: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1180] Device interconnect StreamExecutor with strength 1 edge matrix:
2022-01-19 17:52:57.479391: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1186]      0 
2022-01-19 17:52:57.479414: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1199] 0:   N 
2022-01-19 17:52:57.480982: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1325] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 4399 MB memory) -> physical GPU (device: 0, name: Quadro P2000, pci bus id: 0000:65:00.0, compute capability: 6.1)
Traceback (most recent call last):
  File "legacy/train.py", line 186, in <module>
    tf.app.run()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/platform/app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 312, in run
    _run_main(main, args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 258, in _run_main
    sys.exit(main(argv))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/util/deprecation.py", line 324, in new_func
    return func(*args, **kwargs)
  File "legacy/train.py", line 182, in main
    graph_hook_fn=graph_rewriter_fn)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/object_detection/legacy/trainer.py", line 304, in train
    decent_q.CreateQuantizeTrainingGraph(config=q_config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/quantize_graph.py", line 415, in CreateQuantizeTrainingGraph
    input_graph_def, config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/quantize_graph.py", line 219, in CreateQuantizeTrainingGraphDef
    input_graph_def_string, config.to_string(), status)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/framework/errors_impl.py", line 556, in __exit__
    c_api.TF_GetCode(self.status.status))
tensorflow.python.framework.errors_impl.InvalidArgumentError: Cannot assign a device for operation unstack: Could not satisfy explicit device specification '/device:GPU:0' because no supported kernel for GPU devices is available.
Colocation Debug Info:
Colocation group had the following types and supported devices: 
Root Member(assigned_device_name_index_=-1 requested_device_name_='/device:GPU:0' assigned_device_name_='' resource_device_name_='' supported_device_types_=[CPU] possible_devices_=[]
Unpack: CPU 
Colocation members, user-requested devices, and framework assigned devices, if any:
  unstack (Unpack) /device:GPU:0
Op: Unpack
Node attrs: T=DT_STRING, num=2, axis=0
Registered kernels:
  device='XLA_GPU'; T in [DT_FLOAT, DT_DOUBLE, DT_INT32, DT_UINT8, DT_INT8, ..., DT_BFLOAT16, DT_COMPLEX128, DT_HALF, DT_UINT32, DT_UINT64]
  device='XLA_CPU'; T in [DT_FLOAT, DT_DOUBLE, DT_INT32, DT_UINT8, DT_INT8, ..., DT_BFLOAT16, DT_COMPLEX128, DT_HALF, DT_UINT32, DT_UINT64]
  device='GPU'; T in [DT_INT64]
  device='GPU'; T in [DT_INT32]
  device='GPU'; T in [DT_BOOL]
  device='GPU'; T in [DT_UINT8]
  device='GPU'; T in [DT_BFLOAT16]
  device='GPU'; T in [DT_DOUBLE]
  device='GPU'; T in [DT_FLOAT]
  device='GPU'; T in [DT_HALF]
  device='CPU'; T in [DT_VARIANT]
  device='CPU'; T in [DT_RESOURCE]
  device='CPU'; T in [DT_STRING]
  device='CPU'; T in [DT_BOOL]
  device='CPU'; T in [DT_COMPLEX128]
  device='CPU'; T in [DT_COMPLEX64]
  device='CPU'; T in [DT_DOUBLE]
  device='CPU'; T in [DT_FLOAT]
  device='CPU'; T in [DT_BFLOAT16]
  device='CPU'; T in [DT_HALF]
  device='CPU'; T in [DT_INT8]
  device='CPU'; T in [DT_UINT8]
  device='CPU'; T in [DT_INT16]
  device='CPU'; T in [DT_UINT16]
  device='CPU'; T in [DT_INT32]
  device='CPU'; T in [DT_INT64]
  device='XLA_GPU_JIT'; T in [DT_FLOAT, DT_DOUBLE, DT_INT32, DT_UINT8, DT_INT8, ..., DT_BFLOAT16, DT_COMPLEX128, DT_HALF, DT_UINT32, DT_UINT64]
  device='XLA_CPU_JIT'; T in [DT_FLOAT, DT_DOUBLE, DT_INT32, DT_UINT8, DT_INT8, ..., DT_BFLOAT16, DT_COMPLEX128, DT_HALF, DT_UINT32, DT_UINT64]
     [[{{node unstack}}]]
```

This node **unstack** exists, but it is in the postprocessing layers to build the anchors. 
Can anyone help me, please?

Cheers

xieyi4650, 2022-02-23:
### Config quantize function into `object_detection`
- To quantize finetune model the FixNeuron operation is need to insert to training and evaluation graph. 
  This action is done by graph rewriter builder of `object_detection`. We can use this mechanism to create quantized train/evaluation graph.
  To let the `object_detection` framework use graph rewriter builder. Add code below to the end of `pipeline.config`
  ```
  # in ROOT/ssd_mobilenet_v1_coco_2018_01_28/pipeline.config
  graph_rewriter {
    quantization {}
  }
  ```
- Then edit `ROOT/models/research/object_detection/builders/graph_rewriter_builder.py`,
    - import decent_q in below line 21 by
      ```python
      # in tensorflow/models/research/object_detection/builders/graph_rewriter_builder.py
      from tensorflow.contrib import decent_q
      ```
    - replace function `build` with:
      ```python
      # in tensorflow/models/research/object_detection/builders/graph_rewriter_builder.py
      def build(graph_rewriter_config, is_training):
        """Returns a function that modifies default graph based on options.

        Args:
          graph_rewriter_config: graph_rewriter_pb2.GraphRewriter proto.
          is_training: whether in training of eval mode.
        """
        def decent_q_graph_rewrite_fn():
          """Function to quantize weights and activation of the default graph."""
          # Quantize the graph by inserting quantize ops for weights and activations
          q_config = decent_q.QuantizeConfig(input_nodes=["net_in"],
                                             output_nodes=["Squeeze", "concat_1"],
                                             input_shapes=[[-1, 300, 300, 3]],
                                             weight_bit=8,
                                             activation_bit=8,
                                             method=1,
                                             output_dir="./quantize_results")
          if is_training:
            decent_q.CreateQuantizeTrainingGraph(config=q_config)
          else:
            decent_q.CreateQuantizeEvaluationGraph(config=q_config)
            latest_ckpt = tf.train.latest_checkpoint("./finetune_dir/")
            decent_q.CreateQuantizeDeployGraph(
                                   checkpoint=latest_ckpt,
                                   config=q_config)

          slim.summarize_collection('quant_vars')

        return decent_q_graph_rewrite_fn
      ```

iivanov1996, 2022-07-04:
Hi @xieyi4650,

I wanted to know what kind of PTQ the vai_q_tensorflow1.x performs? Symmetric/Asymmetric? Uniform/Non-Uniform?

Thanks

xieyi4650, 2022-07-27:
@iivanov1996  Symmetric Uniform

hito0512, 2022-10-21:
@samagalhaes hello , where can i find the cases of quantise finetuning ? I am using Vitis AI 1.4 and tf1.15

hito0512, 2022-10-21:
> 
where can i find this file("graph_rewriter_builder.py") and the cases of quantise finetuning ? I am using Vitis AI 1.4 and tf1.15

samagalhaes, 2022-10-21:
It is in the TensorFlow Models package.

> > 
> 
> where can i find this file("graph_rewriter_builder.py") and the cases of quantise finetuning ? I am using Vitis AI 1.4 and tf1.15

JaviMota, 2022-11-16:
Hi 

I implemented what @xieyi4650 suggest in the object_detection and Im getting the following error:

Traceback (most recent call last):
  File "object_detection/model_main.py", line 121, in <module>
    tf.app.run()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/platform/app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 312, in run
    _run_main(main, args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 258, in _run_main
    sys.exit(main(argv))
  File "object_detection/model_main.py", line 117, in main
    tf.estimator.train_and_evaluate(estimator, train_spec, eval_specs[0])
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_estimator/python/estimator/training.py", line 473, in train_and_evaluate
    return executor.run()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_estimator/python/estimator/training.py", line 613, in run
    return self.run_local()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_estimator/python/estimator/training.py", line 714, in run_local
    saving_listeners=saving_listeners)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_estimator/python/estimator/estimator.py", line 370, in train
    loss = self._train_model(input_fn, hooks, saving_listeners)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_estimator/python/estimator/estimator.py", line 1161, in _train_model
    return self._train_model_default(input_fn, hooks, saving_listeners)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_estimator/python/estimator/estimator.py", line 1191, in _train_model_default
    features, labels, ModeKeys.TRAIN, self.config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_estimator/python/estimator/estimator.py", line 1149, in _call_model_fn
    model_fn_results = self._model_fn(features=features, **kwargs)
  File "/workspace/ssd_mobilenetv1/models/research/object_detection/model_lib.py", line 337, in model_fn
    graph_rewriter_fn()
  File "/workspace/ssd_mobilenetv1/models/research/object_detection/builders/graph_rewriter_builder.py", line 38, in decent_q_graph_rewrite_fn
    decent_q.CreateQuantizeTrainingGraph(config=q_config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/quantize_graph.py", line 415, in CreateQuantizeTrainingGraph
    input_graph_def, config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/contrib/decent_q/python/quantize_graph.py", line 219, in CreateQuantizeTrainingGraphDef
    input_graph_def_string, config.to_string(), status)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/framework/errors_impl.py", line 556, in __exit__
    c_api.TF_GetCode(self.status.status))
tensorflow.python.framework.errors_impl.FailedPreconditionError: GetNext() failed because the iterator has not been initialized. Ensure that you have run the initializer operation for this iterator before getting the next element.
	 [[{{node IteratorGetNext}}]]

Do you know how to solve this issue?

Thanks.