Xilinx/Vitis-AI #646: Build the VCK190 Platfrom VitisAI 2.0

Nunigan, 2022-01-24:
Hi,

While running the _make all_, command building the petalinux fails with this error message:

_WARNING: qtbase-5.15.2+gitAUTOINC+40143c189b-r0 do_patch: Fuzz detected:

Applying patch 0002-egl_kms-Modify-the-default-color-format-to-RGB565.patch
patching file src/platformsupport/kmsconvenience/qkmsdevice.cpp
Hunk #1 succeeded at 369 (offset 22 lines).
Hunk #2 succeeded at 393 with fuzz 1 (offset 22 lines).
patching file src/plugins/platforms/eglfs/deviceintegration/eglfs_kms/qeglfskmsgbmcursor.cpp
Hunk #3 succeeded at 302 (offset 20 lines).
patching file src/plugins/platforms/eglfs/deviceintegration/eglfs_kms/qeglfskmsgbmintegration.cpp
Hunk #1 succeeded at 96 (offset -2 lines).
patching file src/plugins/platforms/eglfs/deviceintegration/eglfs_kms_support/qeglfskmsintegration.cpp

The context lines in the patches can be updated with devtool:

    devtool modify qtbase
    devtool finish --force-patch-refresh qtbase <layer_path>

Don't forget to review changes done by devtool!

WARNING: qtbase-5.15.2+gitAUTOINC+40143c189b-r0 do_patch: QA Issue: Patch log indicates that patches do not apply cleanly. [patch-fuzz]
WARNING: unilog-2.0-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.0, attempting MIRRORS if available
ERROR: unilog-2.0-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus,guid=a84f5743b634a00c76947def61ee4cf9"; export SSH_AGENT_PID="7458"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export GIT_SSL_CAINFO="/opt/Xilinx/Petalinux2021.2/components/yocto/buildtools/sysroots/x86_64-petalinux-linux/etc/ssl/certs/ca-certificates.crt"; export PATH="/tmp/xilinx_vck190_base-2021.2/sysroots-uninative/x86_64-linux/usr/bin:/home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/components/yocto/layers/core/scripts:/tmp/xilinx_vck190_base-2021.2/work/cortexa72-cortexa53-xilinx-linux/unilog/2.0-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux:/tmp/xilinx_vck190_base-2021.2/work/cortexa72-cortexa53-xilinx-linux/unilog/2.0-r0/recipe-sysroot/usr/bin/crossscripts:/tmp/xilinx_vck190_base-2021.2/work/cortexa72-cortexa53-xilinx-linux/unilog/2.0-r0/recipe-sysroot-native/usr/sbin:/tmp/xilinx_vck190_base-2021.2/work/cortexa72-cortexa53-xilinx-linux/unilog/2.0-r0/recipe-sysroot-native/usr/bin:/tmp/xilinx_vck190_base-2021.2/work/cortexa72-cortexa53-xilinx-linux/unilog/2.0-r0/recipe-sysroot-native/sbin:/tmp/xilinx_vck190_base-2021.2/work/cortexa72-cortexa53-xilinx-linux/unilog/2.0-r0/recipe-sysroot-native/bin:/home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/components/yocto/layers/core/bitbake/bin:/tmp/xilinx_vck190_base-2021.2/hosttools"; export HOME="/home/nunigan"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror https://gitenterprise.xilinx.com/Vitis/vitis-ai-staging /home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/build/downloads/git2/gitenterprise.xilinx.com.Vitis.vitis-ai-staging --progress failed with exit code 128, output:
Cloning into bare repository '/home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/build/downloads/git2/gitenterprise.xilinx.com.Vitis.vitis-ai-staging'...
fatal: unable to access 'https://gitenterprise.xilinx.com/Vitis/vitis-ai-staging/': Could not resolve host: gitenterprise.xilinx.com

ERROR: unilog-2.0-r0 do_fetch: Fetcher failure for URL: 'git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.0'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /tmp/xilinx_vck190_base-2021.2/work/cortexa72-cortexa53-xilinx-linux/unilog/2.0-r0/temp/log.do_fetch.276325
ERROR: Task (/home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/project-spec/meta-user/recipes-vitis-ai/unilog/unilog_2.0.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3414 tasks of which 1999 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/project-spec/meta-user/recipes-vitis-ai/unilog/unilog_2.0.bb:do_fetch
Summary: There were 3 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build project. Check the /home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/build/build.log file for more details...
make[1]: *** [Makefile:84: images/linux/boot.scr] Error 255
make[1]: Leaving directory '/home/nunigan/Documents/VitsiAI2/Vitis-AI/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd'
make: *** [Makefile:49: linux] Error 2_

dtocci1, 2022-01-24:
I am receiving the same unilog-2.0 BitBake error when building a PetaLinux project with Vitis-AI for the ZCU102 board.

qianglin-xlnx, 2022-01-24:
Hi @Nunigan and @dtocci1 
Currently, the recipes are pointing to internal staging repos. We will update them to public repos soon.

ksstms, 2022-02-02:
Hi @qianglin-xlnx 
When can we expect this to be fixed?

dtocci1, 2022-02-02:
Hi @ksstms, it appears to be fixed as of their push from 5 days ago. I managed to build a Petalinux project with Vitis-AI successfully yesterday, so it looks like they updated the repo links.

ksstms, 2022-02-02:
@dtocci1 Sorry, my bad. I haven't noticed the update

closed 2022-02-10 by Nunigan