Xilinx/Vitis-AI #648: quantize in 3080ti problem

jzymessi, 2022-01-26:
When I use 3080ti to perform quantization in the doctor environment, the GPU shows that the memory is occupied,
but the program has not quantified. 
[image]

[image]

However, in 2080ti, there is no problem in operation. What is the solution?

qianglin-xlnx, 2022-01-26:
Hi @jzymessi 

Nvidia 30xx GPU not supported for Vitis AI Tensorflow 1.x GPU docker. 
You can use other GPU for quantizing TF1 model or use TF2 model

jzymessi, 2022-01-26:
@qianglin-xlnx

jzymessi, 2022-01-26:
@qianglin-xlnx 
thanks，will 30XX GPU be supported in the new version later?

qianglin-xlnx, 2022-01-27:
HI  @jzymessi 

Because CUDA11 is not supported by TF1 itself, we do not intend to support TF1 with 30XX GPU.

closed 2022-02-22 by qianglin-xlnx