Xilinx/Vitis-AI #649: Some problems about quantization accuracy

jzymessi, 2022-01-26:
There are several problems in the process of testing quantization accuracy
A resnet50 model file is downloaded from models zoo. The link is as follows:
 https://www.xilinx.com/bin/public/openDownload?filename=tf_resnetv1_50_imagenet_224_224_6.97G_2.0.zip
According to the display in readme, the accuracy test officially given by Xilinx is as follows:
[image]
Q1:
As shown in the figure above, in the downloaded files, is the file in the quantified folder the model file after running quantization (vai_q)? Is the test accuracy result officially given a model file after quantification (vai_q)? Is this precision consistent with the model precision generated into XModel after executing compilation (vai_c)?
[image]
Q2:
During the compilation (vai_c) of Vitis AI, will there be a loss of accuracy when the model is compressed?
Q3:
If the above quantitative test results cannot represent the final accuracy results, are there other test cases?

lebacuong, 2022-02-21:
Hi guy,

I am deploying my model using vitis-ai pytorch quantizer.The accuracy of the quantized model is quite similar to the original  model(tested both of two models on GPU). But, when i deploy it on FPGA, the accuracy of compiled model is greatly reduced compared to the quantized model. I am not sure how this problem comes from the begginning and how to solve it. Can you guys help me?

Thank all!

xieyi4650, 2022-02-23:
@jzymessi 
A1:yes.  the file in the "quantified" folder the model file after running quantization (vai_q)?
A2:Yes, there will  be a different of accuracy when the model is compressed, but very a bit.

@qianglin-xlnx  maybe answer your question

kavyab11, 2022-02-25:
@lebacuong 

I have been facing the same issue. Were you able to find a solution?

closed 2022-08-19 by jzymessi