Xilinx/Vitis-AI #650: PyTorch Quantizer in VAI 2.0 Docker: TypeError: 'data' must be a numpy ndarray

Hamptonjc, 2022-01-26:
My current quantization setup errors out with the new VAI 2.0 Docker image. It works in the VAI 1.4 Docker image.

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-6-6c11a752c514> in <module>
     48 quantizer = torch_quantizer("test", net,
     49                             input_tensor, output_dir=OUTPUT_DIR, bitwidth=BITWIDTH)
---> 50 quantizer.export_xmodel(output_dir=OUTPUT_DIR)
     51 
     52 

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py in export_xmodel(self, output_dir, deploy_check)
    108   # export xmodel for compilation
    109   def export_xmodel(self, output_dir="quantize_result", deploy_check=False):
--> 110     self.processor.export_xmodel(output_dir, deploy_check)
    111 
    112   # deploy the trained model for quant aware training process

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py in export_xmodel(self, output_dir, deploy_check)
    178   # export xmodel file to be compiled for deployment
    179   def export_xmodel(self, output_dir="quantize_result", deploy_check=False):
--> 180     dump_xmodel(output_dir, deploy_check, self._lstm_app)
    181 
    182 

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py in dump_xmodel(output_dir, deploy_check, lstm_app)
    201 
    202     NndctScreenLogger().info("=>Converting to xmodel ...")
--> 203     deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
    204     #depoly_infos = compiler.get_deloy_graph_infos(quantizer, deploy_graphs)
    205     xmodel_depoly_infos, dump_deploy_infos = compiler.get_xmodel_and_dump_infos(quantizer, deploy_graphs)

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in get_deploy_graph_list(quant_model, nndct_graph)
    288   # sync model data with dev graph
    289   connect_module_with_graph(quant_model, g_optmizer.frozen_graph, recover_param=False)
--> 290   update_nndct_blob_data(quant_model, g_optmizer.frozen_graph)
    291   connect_module_with_graph(quant_model, nndct_graph, recover_param=False)
    292 

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in update_nndct_blob_data(module, graph, time_step)
    163   visited = []
    164   for source in source_nodes:
--> 165     dfs(source, visited)
    166 
    167 

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in dfs(node, visited)
    154     for cn in graph.children(node):
    155       if cn not in visited:
--> 156         dfs(cn, visited)
    157 
    158   source_nodes = []

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in dfs(node, visited)
    154     for cn in graph.children(node):
    155       if cn not in visited:
--> 156         dfs(cn, visited)
    157 
    158   source_nodes = []

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in dfs(node, visited)
    154     for cn in graph.children(node):
    155       if cn not in visited:
--> 156         dfs(cn, visited)
    157 
    158   source_nodes = []

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in dfs(node, visited)
    154     for cn in graph.children(node):
[... 283 lines cut ...]
    158   source_nodes = []

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in dfs(node, visited)
    154     for cn in graph.children(node):
    155       if cn not in visited:
--> 156         dfs(cn, visited)
    157 
    158   source_nodes = []

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py in dfs(node, visited)
    150       in_data = node.in_tensors[0].data
    151       data = permute_data(in_data, node.node_attr(node.op.AttrName.ORDER))
--> 152       node.out_tensors[0].from_ndarray(data)
    153 
    154     for cn in graph.children(node):

/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_tensor.py in from_ndarray(self, data)
     64   def from_ndarray(self, data):
     65     if not isinstance(data, np.ndarray):
---> 66       raise TypeError("'data' must be a numpy ndarray")
     67     self._data = np.copy(data)
     68     self._dtype = self._dtype_map[self._data.dtype]

TypeError: 'data' must be a numpy ndarray
```

niuxjxlnx, 2022-02-21:
Hi, @Hamptonjc:

Can you tell which model are you working on?

niuxjxlnx, 2022-03-01:
It looks like before exporting xmodel, no model evaluation iteration is called. So some node has no data. In Vitis-AI 1.4, there is no check for this situation. But in Vitis-AI 2.0, there is check that data of node must not be None.

closed 2022-03-01 by niuxjxlnx