Xilinx/Vitis-AI #652: YoloV3 takes more than 900 milliseconds with the TVM flow on Jupyter Notebook (Vitis AI 2.0)

corehalt, 2022-01-28:
I am looking for help to find out why YoloV3 is running so slow on the Jupyter notebook demo.
It is giving me more than 900 milliseconds per inference on the notebook.

Setup
   - FPGA card: `Xilinx AlveoU50LV`
   - DPU overlay: `DPUCAHX8H`
   - Stack: `Vitis-AI (master@d02dcb6041663dbc7ecbc0c6af9fafa087a789de)`
     - TVM target: `DPUCAHX8H-u50lv`

I tested the setup with other networks to validate the installation:

## Resnet-18
### Compiled with `external/tvm/examples/compile_mxnet_resnet_18.py`)
```
(vitis-ai-tensorflow) Vitis-AI /workspace/external/tvm/examples > python3 run_mxnet_resnet_18.py -f tvm_dpu_cpu.so                                                                                                                                                                                                           
WARNING: Logging before InitGoogleLogging() is written to STDERR                                                                                                                                                                                                                                                             
I0128 01:31:20.014504  1803 dpu_func.cpp:65] create running for subgraph: subgraph_eltwise-94575482209408                                                                                                                                                                                                                    
========================================                                                                                                                                                                                                                                                                                     
TVM prediction top-1: 285 Egyptian cat                                                                                                                                                                                                                                                                                       
========================================                                                                                                                                                                                                                                                                                     
========================================                                                                                                                                                                                                                                                                                     
Inference time: 9.37 ms                                                                                                                                                                                                                                                                                                      
========================================                                                                                                                                                                                                                                                                                     
```

## Resnet-50
### Compiled with *modified* script `external/tvm/examples/compile_mxnet_resnet_18.py`)
(Instead of downloading the ResNet18 model we download the ResNet50 model)
```
(vitis-ai-tensorflow) Vitis-AI /workspace/external/tvm/examples > python3 run_mxnet_resnet_18.py -f tvm_dpu_cpu.so                                                                                                                                                                                                           
WARNING: Logging before InitGoogleLogging() is written to STDERR                                                                                                                                                                                                                                                             
I0128 02:03:08.526906  2355 dpu_func.cpp:65] create running for subgraph: subgraph_eltwise-93889653737856                                                                                                                                                                                                                    
========================================                                                                                                                                                                                                                                                                                     
TVM prediction top-1: 285 Egyptian cat                                                                                                                                                                                                                                                                                       
========================================                                                                                                                                                                                                                                                                                     
========================================                                                                                                                                                                                                                                                                                     
Inference time: 14.99 ms                                                                                                                                                                                                                                                                                                     
========================================                                                                                                                                                                                                                                                                                                                                                                                                               
```

## YoloV3

Then I run the Jupyter notebook https://github.com/Xilinx/Vitis-AI/blob/master/external/tvm/examples/external_yolov3_tutorial.ipynb
Selected the `TF` TVM flow and the TVM target `DPUCAHX8H-u50lv`.
This saves the compiled model to a shared library that later I can run using the provided script:

```
(vitis-ai-tensorflow) Vitis-AI /workspace/external/tvm/examples > python run_external_yolov3.py -f tvm_dpu_cpu.so 
[ WARN:0@1.612] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (239) findDecoder imread_('/home/root/tensorflow-yolov3/docs/images/road.jpeg'): can't open/read file: check file path/integrity
Loading the compiled model ...
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0128 02:45:19.429919  3521 dpu_func.cpp:65] create running for subgraph: subgraph_concat-94524824654544
Running the compiled model ...
========================================
Detection output stored in ./output.png
========================================
Inference time: 58.42 ms
========================================
```
But then if I measure the time from the Jupyter notebook it gives me more than 900 milliseconds per inference.
I can share a PDF that shows all the options I chose.
[external_yolov3_tutorial - Jupyter Notebook.pdf](https://github.com/Xilinx/Vitis-AI/files/7957589/external_yolov3_tutorial.-.Jupyter.Notebook.pdf)

Did I make a mistake on the Jupyter notebook? Why the latencies doesn't match?

Thank you

anilmartha, 2022-01-28:
@corehalt 

I am able to see same latency across standalone python script and jupyter.  Your Jupiter steps seems to be correct except setting environment variable PX_QUANT_SIZE. Could you try without setting env variable PX_QUANT_SIZE in Jupiter notebook? 
Could you also double-check that the image is being read correctly in the standalone script? I see the **warning [ WARN:0@1.612] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (239) findDecoder imread_('/home/root/tensorflow-yolov3/docs/images/road.jpeg'): can't open/read file: check file path/integrity**  in your log. 

Below are my running logs in standalone and Jupiter. 
**standalone script** 
[image]
**Jupyter**
[image]

SuZhan0322, 2022-09-01:
Hi @corehalt , It looks like your issue has been solved, please reopen this issue if you have any problems. thanks，

closed 2022-09-01 by SuZhan0322