Xilinx/Vitis-AI #653: Can not find PyTorch and Tensorflow2 Optimizer Tools  in VAI 2.0 Docker:

jia0511, 2022-01-28:
Hello xilinx,

Judging from the vitis-ai2.0 information you released as follow, it can be seen that TF2 and pytorch pruning tools are currently supported, but after I entered, there is only the vaitrace tool, not the vai_p tool. Did you miss it or do you need other configurations? 

Optimizer
TensorFlow 1.15
Support tf.keras.Optimizer for model training
TensorFlow 2.x
Support TensorFlow 2.3-2.6
Add iterative pruning
PyTorch
Support PyTorch 1.4-1.9.1
Support shared parameters in pruning
Add one-step pruning
Add once-for-all(OFA)
Unified APIs for iterative and one-step pruning
Enable pruned model to be used by quantizer
Support nn.Conv3d and nn.ConvTranspose3d

gau-nernst, 2022-01-29:
I think the pre-built CPU image does not include optimizer tools. You need to build your on GPU docker image for those.

jia0511, 2022-01-31:
My problem occurred in the GPU version, and the GPU version built by other colleagues has this problem, and the vai_p_tensorflow2 or vai_p_pytorch tools cannot be found.

white0dew, 2022-02-08:
> My problem occurred in the GPU version, and the GPU version built by other colleagues has this problem, and the vai_p_tensorflow2 or vai_p_pytorch tools cannot be found.

hello,would pls teach me how to download the GPU version built by other colleagues? 
Thank you!

jia0511, 2022-02-09:
Dear everyone,

I found that this interface does not exist in the GPU version in binary form. pytorch is encapsulated through the nndct python interface, and the same is true for tf2. The example is in the tools/Vitis-AI-Optimizer directory, so I close this ticket. 
Thank you!