Xilinx/Vitis-AI #655: Symmetric padding for PyTorch model

gau-nernst, 2022-01-29:
Hello,

I want to use symmetric padding for my PyTorch model, since DPU can only use symmetric padding. Since there is no symmetric padding in PyTorch, there is no mapping between PyTorch modules and DPU instructions. Is there a way for me to insert DPU symmetric padding instruction in the xmodel after I convert PyTorch model to xmodel?

Thanks

niuxjxlnx, 2022-02-24:
Hi, @gau-nernst:

DPU supports not only symmetric padding but also constant padding. You can try constant padding.

gau-nernst, 2022-06-16:
No. In the end I did that part on CPU, since it was easier to write the code.

masrourabdou, 2022-06-16:
I have a missunderstand you mean that you write plugin and pass through compilation but you must layer be quantized

closed 2022-06-16 by niuxjxlnx