Xilinx/Vitis-AI #656: vai_q_tensorflow compiler gives error : AssertionError: [ERROR] TF Conv2d requires two inputs: actual: ['input_data']

Afef00, 2022-01-29:
I ran Vitis AI CPU docker container from host using ./docker_run.sh xilinx/vitis-ai-cpu:1.3.411. The notebook runs fine, until the last part when I tried to run vai_q_tensorflow compiler  as below:
```

!vai_c_tensorflow \
    --frozen_pb custom_dense1.pb \
    --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json \
    --output_dir . \
    --net_name mnist_dense1_modif

```
I got the following message error

VITIS_AI Compilation - Xilinx Inc.
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['custom_dense1.pb'], model_type='tensorflow', out_filename='./mnist_dense1_modif_org.xmodel', proto=None)
[INFO] tensorflow model: custom_dense1.pb
[INFO] parse raw model : 8%| | 1/12 [00:00<00:00, 3675.99it/s]
Traceback (most recent call last):
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in
sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/main.py", line 194, in main
normal_run(args)
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/main.py", line 178, in normal_run
in_shapes=in_shapes if len(in_shapes) > 0 else None,
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
model_files, layout, in_shapes=in_shapes, model_type=model_t
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 243, in __create_xmodel_from_tf1
xmodel_name, layout, layers, const_layer_dict, super_const_dict, in_shapes
File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 1847, in __generate_xmodel
), f"[ERROR] TF Conv2d requires two inputs: actual: {bottom}."
AssertionError: [ERROR] TF Conv2d requires two inputs: actual: ['input_data'].

Any ideas how to solve this problem please?
Thanks!

zhenzhen-Xilinx, 2022-03-07:
Hello,Sorry for late reply.Please provide model files.Thanks.

zhenzhen-Xilinx, 2022-03-07:
Can you share your float model and quantized model so we can analyze the problem further. Thank you.

Afef00, 2022-04-04:
@zhenzhen-Xilinx  thank you for the reply!
Actually, I tried with another model but still have a problem.
After the training of my model (ResNet50 for MNIST classification) which is saved as (tf1_classifier.h5), I freeze this model using freeze_graph.py script. Then I run the quantization to obtain a .pb model (deploy_model .pb). After that, I truncated the quantized model to keep only the CONV+POOL stage. I wanted to run this truncated model (Resnet_flatten.pb) using !vai_c_tensorflow \ however I got the following error
`**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['Resnet_flatten.pb'], model_type='tensorflow', out_filename='./Resnet_flatten_org.xmodel', proto=None)
[INFO] tensorflow model: Resnet_flatten.pb
[INFO] parse raw model     :  1%| | 1/124 [00:00<00:00, 2837.82it/s]            
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 243, in __create_xmodel_from_tf1
    xmodel_name, layout, layers, const_layer_dict, super_const_dict, in_shapes
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 1931, in __generate_xmodel
    ), f"[ERROR] tf pad op requires two inputs: actual: {len(bottom)}"
AssertionError: [ERROR] tf pad op requires two inputs: actual: 1`
Any help please!

zhenzhen-Xilinx, 2022-04-06:
Hi, @Afef00. Please send the float and quantized model to my email zhenzhen@xilinx.com. Thanks.

Afef00, 2022-04-07:
Ok thanks!

zhenzhen-Xilinx, 2022-04-28:
Hi @Afef00,

I analyzed the model you provided,this error is due to wrong instruction.

First, the floating-point model needs to be quantized. The reference instructions are as follows:
- ` vai_q_tensorflow quantize --input_frozen_graph freeze_graph.pb --input_nodes input_1 --output_nodes dense/Softmax --input_shapes ?,32,32,3 --input_fn input_fn.calib_input`
After the quantization instruction runs successfully, the model "quantize_eval_model.pb" will be obtained.

Then compile the quantized model. The reference instructions are as follows:
- `vai_c_tensorflow -f quantize_eval_model.pb -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json -o ./ -n skinlsom --options '{"input_shape": "1,32,32,3"} '`

Afef00, 2022-04-28:
Ok, thank you for the help!

zhenzhen-Xilinx, 2022-05-07:
You are welcome. I'm going to close this issues. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-05-07 by qianglin-xlnx