Xilinx/Vitis-AI #659: Compilation failure of YOLOv4

wwMark, 2022-01-30:
Hi all,
I have encounted an unknown issue when compiling the YOLOv4 model. I have used the model [here](https://github.com/RobotEdh/Yolov-4) and follow the [hackster tutorial ](https://www.hackster.io/AlbertaBeef/vitis-ai-1-4-flow-for-avnet-vitis-platforms-5ebc3f#comments) to compile it. The quantization has finished successfully, while the following issue occurs during compilation:

(vitis-ai-tensorflow2) Vitis-AI ~ > vai_c_tensorflow2 -m quantized_yolov4_model.h5 -a custom.json -o ./compiled_yolov4_model -n yolov4
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_yolov4_model.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/yolov4_org.xmodel', proto=None)
[INFO] tensorflow2 model: /home/vitis-ai-user/quantized_yolov4_model.h5
[INFO] keras version: 2.4.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|███████| 264/264 [00:00<00:00, 28644.13it/s]             
[INFO] infer shape (NHWC)  :100%|███████| 519/519 [00:00<00:00, 570.65it/s]               
[INFO] perform level-0 opt :100%|███████| 2/2 [00:00<00:00, 35.65it/s]                    
[INFO] perform level-1 opt :100%|███████| 2/2 [00:00<00:00, 146.92it/s]                   
[INFO] generate xmodel     :100%|███████| 519/519 [00:00<00:00, 865.39it/s]               
[INFO] dump xmodel: /tmp/yolov4_org.xmodel
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Graph name: Yolo_v4, with op num: 959
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = quant_layer_112, type = pool-fix} has been assigned to CPU: ["kernel_height(13) is not in DPU supported range [1, 2]].
[UNILOG][WARNING] xir::Op{name = quant_layer_110, type = pool-fix} has been assigned to CPU: ["kernel_height(9) is not in DPU supported range [1, 2]].
module_infer = 0, counter_m[module_infer] = 12, counter_p[module_infer] = 12
module_idx = 0, counter_m[module_idx] = 12, counter_p[module_idx] = 12
module_idx = 1, counter_m[module_idx] = 0, counter_p[module_idx] = 0
module_idx = 2, counter_m[module_idx] = 0, counter_p[module_idx] = 0
module_idx = 3, counter_m[module_idx] = 0, counter_p[module_idx] = 0
[UNILOG][FATAL][XCOM_PM_FAIL][The compiler occurs an error when generating instructions, please contact us.] 
*** Check failure stack trace: ***
Aborted (core dumped)

My configuration is as follows:
OS: Ubuntu 18.04
VitisAI docker image version: xilinx/vitis-ai:1.4.916

Thanks in advance :)

wanghong4compiler, 2022-02-17:
Hi @wwMark ,
Could you send out the model file "quantized_yolov4_model.h5", so that I can reproduce this issue?

qianglin-xlnx, 2022-03-23:
Hi @wwMark 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-23 by qianglin-xlnx