Xilinx/Vitis-AI #665: Anyone experienced in quantizing and compiling SlowFast-Network with Vitis AI?

MarvS95, 2022-02-03:
Hello,

I am trying to quantize a network for human action recognition (SLOWFAST_8x8_R50 from open source codebase PySlowFast) using Vitis AI 2.0. SlowFast is a pytorch model which has two pathways based on resnet50. I customized the quantization sample (resnet18_quant.py) and tried to run it, but the quantization fails after tracing the model.
[image]

There is an error "TypeError: reconnect_nodes() takes 1 positional argument but 2 were given". Does anyone know how to solve this problem?

Many thanks!

niuxjxlnx, 2022-02-09:
Hi, @MarvS95:

This is a bug in vai_q_pytorch code, you can bypass it by change this file /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/opt_pass.py, line 332, to:

raw_graph.reconnect_nodes()

MarvS95, 2022-02-09:
Hi @niuxjxlnx 

Thank you really much! I removed the argument a few days ago because reconnect_nodes() has no argument in other usages. 

At the moment I'm struggling with the model input in the evaluate() function (customized resnet18_quant.py).

**slowfast_quant.py:**

```
def evaluate(model, val_loader, loss_fn):
  . . .
  for iteraction, (images, labels) in tqdm(
      enumerate(val_loader), total=len(val_loader)):
    images = images.to(device)
    labels = labels.to(device)
    outputs = model(images)
  . . .
```

The SlowFast network needs two tensors as input - one for each of the two pathways. In the script which runs the network on my gpu the input is given as a list of two tensors. When I try to give the model a list of two tensors as input in the evaluate() function, I get errors like the following in several pytorch_nndct scripts.

_AttributeError: 'list' object has no attribute 'device'_ 

How can I give the model two tensors as input in evaluate() without a list?

Thank you really much for your help.

MarvS95, 2022-02-11:
Hi @Alexiazzf 

thank you really much for your reply. I tried your solution but giving two pytorch tensor objects unfortunately also didn't work:
```
slow_pathway = torch.randn([1, 3, 8, 256, 256])
fast_pathway = torch.randn([1, 3, 32, 256, 256])
outputs = model(slow_pathway, fast_pathway)
```
Error Message:

```
free(): invalid next size (fast)
Aborted (core dumped)
```
Did I implement your idea correctly? Today I will also try to concatenate both input tensors for the quantizer and split them up later.

Alexiazzf, 2022-02-14:
Hi @MarvS95 ，

Have you change your original model input at same format? For multi-input model, there are several steps different:

1. Your original model should have two input tensor (here is _x_ and _y_):
`class Net(tnn.Module):
    def __init__(self):
        super(Net, self).__init__()
     
    def forward(self, x: torch.Tensor, y: torch.Tensor):
        out_tensor = x + y
        return out_tensor`
2. When you do quantize, you should put two tensors into one tuple as input:
`input1 = torch.randint(0, 256, (1, args.channels, args.height, args.width)).type(torch.FloatTensor)
    input2 = torch.randint(0, 256, (1, args.channels, args.height, args.width)).type(torch.FloatTensor)
    input = (input1, input2)
quantizer = torch_quantizer(quant_mode, model, input)`
3. When you evaluate your model, input two tensors:
`out_tensor = model(image1, image2)`

MarvS95, 2022-02-14:
Hi @Alexiazzf ,

thank you for the example. What do you mean with same format for original model input? Do you mean that I should change the 2nd dimension of the input (number of frames) to the same number for both tensors?

```
slow_pathway = torch.randn([1, 3, 32, 256, 256])
fast_pathway = torch.randn([1, 3, 32, 256, 256])
```
I changed the model input as you described in the forward-function of the model definition and put a tuple of input tensors to the quantizer, but I still get an error:

**model definition:** 

```
    def forward(self, x: torch.Tensor, y: torch.Tensor):
        m = getattr(self, "pathway{}_stem".format(0))
        x[0] = m(x)
        x[1] = m(y)
        return x
```

**quantize()-function:**

```
slow_pathway = torch.randn([batch_size, 3, 32, 256, 256])
fast_pathway = torch.randn([batch_size, 3, 32, 256, 256])
inputs = (slow_pathway, fast_pathway)

quantizer = torch_quantizer(quant_mode, model, (inputs), output_dir="SLOWFAST_8x8_R50/vai_q_output")
quant_model = quantizer.quant_model
```

**error message:**

```
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 709, in _slow_forward
    result = self.forward(*input, **kwargs)
TypeError: forward() takes 2 positional arguments but 3 were given
```

Alexiazzf, 2022-02-14:
Hi @MarvS95 ,

Sorry, maybe I did not say it clearly. It does not need to "change the2nd dimension of the input (number of frames) to the same number for both tensors". "Same format" mean you should input two tensors in both `forward(self, x, y)` and `model(x, y)` where `x` and `y` are torch.Tensor.

Error message you have got maybe because your model still just have one input tensor, but when you evaluate, you input two tensors.

I want to show you another example:

Wrong code:

```python
import torch
class Net(torch.nn.Module):
    def __init__(self):
        super(Net, self).__init__()
        self.linear = torch.nn.Linear(5, 5)

    def forward(self, input):
        x = input[0]
        y = input[1]
        output = torch.tanh(self.linear(x) + y)
        return output

net = Net()
x, y = torch.rand(4, 5), torch.rand(4,5)
print(net(x,y))
```

Right code:
```python
import torch
class Net(torch.nn.Module):
    def __init__(self):
        super(Net, self).__init__()
        self.linear = torch.nn.Linear(5, 5)

    def forward(self, x, y):
        output = torch.tanh(self.linear(x) + y)
        return output 

net = Net()
x, y = torch.rand(4, 5), torch.rand(4, 5)
print(net(x, y))
```

You can just run these codes. Hope it can help you with your problem.

MarvS95, 2022-02-15:
Hi @Alexiazzf 

thank you for the example code. Now I understand where the error message comes from. And now I also got what you mean with the same format for model() and forward().

The problem yesterday was that I only changed one forward() function. The SlowFastModel starts with the VideoModelStem, which consists of two ResNetBasicStems. Therefore, the SlowFastModel's forward() function calls the VideoModelStem's forward() function. Then the VideoModelStem calls the ResNetBasicStem's forward() function twice (once for pathway0 and once for pathway1). At the end both paths are fused by a fourth forward() function:

**model:**

```
SlowFastModel(
  (s1): VideoModelStem(
    (pathway0_stem): ResNetBasicStem(...)
    (pathway1_stem): ResNetBasicStem(...)
  )
  (s1_fuse): FuseFastToSlow(...)
```
**forward() functions:**

```
    #   SlowFastModel:

    def forward(self, x):
        x = self.s1(x)         #  here VideoModelStem gets built
        x = self.s1_fuse(x)
                (...)
        x = self.s5(x)
        x = self.head(x)
        return x

    #	VideoModelStem:
	
    def forward(self, x):
        for pathway in range(len(x)):
            m = getattr(self, "pathway{}_stem".format(pathway))         #  here both ResnetBasicStem get built
            x[pathway] = m(x[pathway])
        return x

    #	ResnetBasicStem (gets called two times):

    def forward(self, x):
        x = self.conv(x)
        x = self.bn(x)
        x = self.relu(x)
        x = self.pool_layer(x)
        return x

    #	fusion:

    def forward(self, x):
        x_s = x[0]
        x_f = x[1]
        fuse = self.conv_f2s(x_f)
        fuse = self.bn(fuse)
        fuse = self.relu(fuse)
        x_s_fuse = torch.cat([x_s, fuse], 1)
        return [x_s_fuse, x_f]
```

I will modify them later and tell you if it works.

Thanks

MarvS95, 2022-02-17:
Hello @Alexiazzf 

I think I need some more help. Trying to modify the input for two pathways, I got the following error message:
```

  File "/workspace/SlowFast_Quantizer/stem_helper.py", line 98, in forward
    x[pathway] = m(x)
RuntimeError: The expanded size of the tensor (256) must match the existing size (64) at non-singleton dimension 3.  Target sizes: [3, 8, 256, 256].  Tensor sizes: [64, 8, 64, 64]
```

I changed the model input and two of the forward()-functions. In line 3 of the forward()-function of the hole model, I added a second input "y":

**SlowFastModel:**
```
    def forward(self, x: torch.Tensor, y: torch.Tensor):  # <== here I added second input y
        x = self.s1(x, y)  # <== here I added second input y
        x = self.s1_fuse(x)
        x = self.s2(x)
        ( . . . )
        x = self.head(x)
        return x
```
Then I modified forward() of the VideoModelStem for two inputs. Both Tensors come from line 3 in forward() above.

**VideoModelStem:**
```
    def forward(self, x: torch.Tensor, y: torch.Tensor):
        for pathway in range(len(x)+len(y)):
            if pathway == 0:
                m = getattr(self, "pathway{}_stem".format(pathway))
                x[pathway] = m(x)
            if pathway == 1:
                m = getattr(self, "pathway{}_stem".format(pathway))
                x[pathway] = m(y)
        return x
```

This is the input:

```
    quant_stub1 = torch.randn([1, 3, 8, 256, 256])
    quant_stub2 = torch.randn([1, 3, 32, 256, 256])
    quant_stub = (quant_stub1, quant_stub2)
```

Do you know what I can do to fix this?

Thank you

Alexiazzf, 2022-02-18:
Hi @MarvS95 ,

Your Problem of "RuntimeError: The expanded size of the tensor (256) must match the existing size (64) at non-singleton dimension 3.  Target sizes: [3, 8, 256, 256].  Tensor sizes: [64, 8, 64, 64]" is maybe because the size of tensors is wrong. You should check your model and data for right tensor size.

This error is about pytorch functions. You can read pytorch document for finding more details. https://pytorch.org/docs/stable/torch.html

MarvS95, 2022-02-18:
Hi @Alexiazzf,

thanks for the advice. Using format(x.shape) helped me. I made an indexing mistake, so forward() didn't get the right tensor size.

closed 2022-02-21 by niuxjxlnx