Xilinx/Vitis-AI #666: Error when creating new conda environment for PyTorch using replace_pytorch.sh script

daniperfer, 2022-02-03:
Hi:

The Vitis-AI Quantizer documentation (https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#quick-start-in-docker-environment) says that a new conda environment for PyTorch can be created by using the script `/opt/vitis_ai/scripts/replace_pytorch.sh`.

> there is a conda environment "vitis-ai-pytorch", in which vai_q_pytorch package is already installed. In this conda environment, python version is 3.6, pytorch version is 1.7.1 and torchvision version is 0.8.2. [...] A new Conda environment with a specified PyTorch version (1.2~1.9) can be created using the /opt/vitis_ai/scripts/replace_pytorch.sh script. This script clones a Conda environment from vitis-ai-pytorch, uninstalls the original PyTorch, Torchvision and vai_q_pytorch packages, and then installs the specified version of PyTorch, Torchvision, and re-installs vai_q_pytorch from source code.

I'd used `replace_pytorch.sh` script in Vitis v1.4.1 for creating a new conda environment and it worked well. 

`bash /opt/vitis_ai/scripts/replace_pytorch.sh my-vitis-ai-pytorch`

However, when I've tried to do the same in Vitis-AI v2.0, I've got the following errors:
```
PathNotFoundError: /scratch/conda-channel/linux-64/unilog-2.0.0-hea4fdf2_87.tar.bz2 
PathNotFoundError: /scratch/conda-channel/linux-64/target_factory-2.0.0-h680af44_97.tar.bz2
PathNotFoundError: /scratch/conda-channel/linux-64/xir-2.0.0-py36h893bffd_109.tar.bz2
PathNotFoundError: /scratch/conda-channel/linux-64/dpuv3-pycompiler-2.0-py36h3f44cd5_101.tar.bz2
PathNotFoundError: /scratch/conda-channel/linux-64/pytorch_nndct-2.0.0-py36h164a702_85.tar.bz2
PathNotFoundError: /scratch/conda-channel/linux-64/xcompiler-2.0.0-py36hea4fdf2_133.tar.bz2
PathNotFoundError: /scratch/conda-channel/linux-64/xnnc-2.0.0-py36hd51751d_109.tar.bz2
PathNotFoundError: /scratch/conda-channel/linux-64/vaic-2.0.0-py36hd51751d_44.tar.bz2

#### New conda environment is NOT created correctly.
```

And the new environment is not created correctly. Have anybody experienced the same problem? 

Do someone know if there is a way for correctly creating a new conda pytorch env with that script?

niuxjxlnx, 2022-02-22:
Hi, @daniperfer:

A work round to solve the problem, insert the following lines into replace_pytorch.sh before creating the new conda environment: 
```
echo -e "\n#### Creating a new conda environment by cloning vitis-ai-pytorch and activate it..."

**sudo chmod 777 /opt/vitis_ai/conda
cd /scratch/
wget -O conda-channel.tar.gz --progress=dot:mega https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_2.0.0.1103-02.tar.gz
tar -xzvf conda-channel.tar.gz
source /opt/vitis_ai/conda/etc/profile.d/conda.sh**

conda create -n $1 --clone vitis-ai-pytorch 
```

niuxjxlnx, 2022-02-23:
Hi, @daniperfer:

Please try this version of script. 
[replace_pytorch.zip](https://github.com/Xilinx/Vitis-AI/files/8124601/replace_pytorch.zip)

daniperfer, 2022-02-24:
Thanks @niuxjxlnx.
With this version I could create a new conda environment for PyTorch.

closed 2022-02-25 by niuxjxlnx