Xilinx/Vitis-AI #667: The 2.0 Vitis AI release tarball includes precompiled binary files (.so, .o, executables)

sil2100, 2022-02-03:
When packaging 2.0 for Ubuntu, I noticed that the https://github.com/Xilinx/Vitis-AI/releases/tag/v2.0 tarball includes 59 binary precompiled files weighting an extra 28MB. Are those necessary in the source? Or have those been included by mistake?

As one can never be sure what those binaries are and how those were built, I am wondering if they should be in the release tarball. Most of those seem like left-over build artifacts, but maybe some of them are actually useful? Would be good to know as if any of those binaries are required for operation, we might need to demote the Ubuntu vitis-ai package to multiverse due to licensing.

Thank you!

List of found binary files follows:
`
./dsa/DPU-TRD/app/samples/lib/libvart-buffer-object.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-xrt-device-handle.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libxir.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libunilog.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-runner.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-mem-manager.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-dpu-runner.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-dpu-controller.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-util.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-trace.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-runner-assistant.so.2.0.0
./dsa/DPU-TRD/app/samples/bin/resnet50
./demo/DPU-for-RNN/rnnt_asr_vck5000/model/transformer
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/test
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/xrnn_py.so
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/test.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CompilerIdCXX/a.out
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CMakeDetermineCompilerABI_C.bin
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CompilerIdC/a.out
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CMakeDetermineCompilerABI_CXX.bin
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/python/xrnn_python.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/test
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/xrnn_py.so
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/libxrnn.so
./tools/Vitis-AI-Library/graph_runner/samples/resnet50_graph_runner/resnet50_graph_runner
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CompilerIdCXX/a.out
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CMakeDetermineCompilerABI_C.bin
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CompilerIdC/a.out
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CMakeDetermineCompilerABI_CXX.bin
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/main.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/helper.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/anchor.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/parse_display_result.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/pointpillars_post.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/preprocess.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/train.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/voxel_generator.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/box_coder.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/target.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/optimizer.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/preprocess.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/input_reader.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/model.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/anchors.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/activations.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/pipeline.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/losses.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/sampler.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/layers.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/second.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/similarity.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/sample_pointpillars_graph_runner
./tools/Vitis-AI-Quantizer/vai_q_tensorflow1.x/tensorflow/compiler/aot/codegen_test_o.golden
`

qianglin-xlnx, 2022-02-08:
HI @sil2100 
The following files are not necessary in the release tarball. We will remove them.
```

./tools/Vitis-AI-Library/graph_runner/samples/resnet50_graph_runner/resnet50_graph_runner
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CompilerIdCXX/a.out
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CMakeDetermineCompilerABI_C.bin
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CompilerIdC/a.out
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/3.16.0/CMakeDetermineCompilerABI_CXX.bin
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/main.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/helper.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/anchor.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/parse_display_result.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/pointpillars_post.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/src/preprocess.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/train.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/voxel_generator.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/box_coder.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/target.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/optimizer.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/preprocess.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/input_reader.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/model.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/anchors.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/activations.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/pipeline.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/losses.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/sampler.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/layers.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/second.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/CMakeFiles/sample_pointpillars_graph_runner.dir/second/protos/similarity.pb.o
./tools/Vitis-AI-Library/graph_runner/samples/pointpillars_graph_runner/build/sample_pointpillars_graph_runner
```

For the following files, they are used for DPU-TRD sample test. I think you can ignore or remove them.

```
./dsa/DPU-TRD/app/samples/lib/libvart-buffer-object.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-xrt-device-handle.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libxir.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libunilog.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-runner.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-mem-manager.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-dpu-runner.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-dpu-controller.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-util.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-trace.so.2.0.0
./dsa/DPU-TRD/app/samples/lib/libvart-runner-assistant.so.2.0.0
./dsa/DPU-TRD/app/samples/bin/resnet50
```
The following files are for RNN. I think you can also ignore them if you don't run RNN on VCK5000. 

```
./demo/DPU-for-RNN/rnnt_asr_vck5000/model/transformer
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/test
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/xrnn_py.so
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/test.cpp.o
[... cut ...]
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/python/xrnn_python.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/test
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/xrnn_py.so
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/libxrnn.so
```

qianglin-xlnx, 2022-02-17:
Hi @sil2100 
I confirm that the following files can be removed. 

```
./demo/DPU-for-RNN/rnnt_asr_vck5000/model/transformer
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/test
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/xrnn_py.so
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/test.dir/test.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CompilerIdCXX/a.out
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CMakeDetermineCompilerABI_C.bin
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CompilerIdC/a.out
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/3.18.0/CMakeDetermineCompilerABI_CXX.bin
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn_py.dir/python/xrnn_python.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn.dir/src/lstm_xrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/CMakeFiles/xrnn.dir/src/xxrnn.cpp.o
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/test
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/xrnn_py.so
./demo/DPU-for-RNN/rnnt_asr_vck5000/hwlib/libxvrnn/build/libxrnn.so
```

closed 2022-02-25 by qianglin-xlnx