Xilinx/Vitis-AI #671: about the pytorch add!!!!!!!! what's wrong?!!!

white0dew, 2022-02-09:
when i tested this model:
[image]

the process of calib is ok.

but it occured in the "test --deploy":
[image]

This is confused me very very bad,  how can i correct it?
I ony want to find a way to implement the hardsigmoid , pls help me ,Thank you very much!

Alexiazzf, 2022-02-22:
Hi @white0dew ,

Which version of Vitis-AI you used? I tested your example and it can deploy well.

white0dew, 2022-02-22:
> 

Hi @Alexiazzf 
My Vitis-AI is 1.4, could you upload your deployed xmodel?
I had tested Vitis-AI 2.0 , and I will get segmentation when I test any model.

Alexiazzf, 2022-02-22:
Hi @white0dew ,

Here is my xmodel:
[Model_int (5).zip](https://github.com/Xilinx/Vitis-AI/files/8115184/Model_int.5.zip)

white0dew, 2022-02-22:
Hi @Alexiazzf 
Which version of Vitis-AI you used? 
Now I am very confused....

Alexiazzf, 2022-02-22:
Hi @white0dew ,

I use Vitis-AI 2.0.

white0dew, 2022-02-22:
Thanks @Alexiazzf 
I will test the Vitis-AI tomorrow again, your Docker is GPU version?

Alexiazzf, 2022-02-23:
Hi @white0dew ,

Yes, my docker is GPU version.

Alexiazzf, 2022-02-25:
Hi @white0dew ,

Have you solved your problem? In Vitis-AI 1.4, you can change your scalar “1” in `torch.add(x, 1)` to a Tensor. There is a bug in Vitis-AI1.4 when a tensor add a scalar directly. And it has been fixed in Vitis-AI 2.0.

white0dew, 2022-02-25:
> Hi @white0dew ,
> 
> Have you solved your problem? In Vitis-AI 1.4, you can change your scalar “1” in `torch.add(x, 1)` to a Tensor. There is a bug in Vitis-AI1.4 when a tensor add a scalar directly. And it has been fixed in Vitis-AI 2.0.
> 
> When your problem have solved, please let me know, thanks.

Very Sad, when I used the Vitis-Ai 2.0 there is a problem which confused me very much.
[image]

No matter I quant which model ， this problem always occur......
Can you help me?

white0dew, 2022-02-25:
And My Vitis-Ai image is CPU version ,which is downloaded from your docker hub.

Alexiazzf, 2022-02-25:
Hi @white0dew,

I think "segmentation fault" is not caused by quantization of `torch.add(x, 1)`. What kind of CPU and GPUs are you using?

white0dew, 2022-02-25:
> Hi @white0dew,
> 
> I think "segmentation fault" is not caused by quantization of `torch.add(x, 1)`. What kind of CPU and GPUs are you using? And what kind of operating system are you using? And which version? Do you use original conda environment in docker? Or you made your own conda environment in Vitis-AI docker?

Thanks! I solve this problem which is caued by tensot.to(cuda) .
PS:Beacuse My version is cpu...

But when i compile model to KV260 , i find that the add op is run on cpu, why?

So there is no way for me to implement a hard_sigmoid op on KV260?

Alexiazzf, 2022-02-25:
Hi @white0dew ,

Only VCK190 support HardSigmoid compiling. Other hardware will transfer HardSigmoid to CPU automatically.

white0dew, 2022-02-25:
> Hi @white0dew ,
> 
> Only VCK190 support HardSigmoid compiling. Other hardware will transfer HardSigmoid to CPU automatically.

Thank you very much , then  i consider use another op to replace the h_sigmoid.

closed 2022-02-25 by niuxjxlnx