Xilinx/Vitis-AI #672: Versal DPU B64 performance

ksstms, 2022-02-09:
I was very excited to see the release of Vitis-AI 2.0, especially the new DPUCVDX8G variants that double the number of AIE cores used per batch. I benchmarked them using XVDPU-TRD on a VCK190-ES board, and unfortunately the performance of the C64 variants is underwhelming.

I used `cf_resnet50_imagenet_224_224_7.7G` for the tests. I got 1436 FPS on C32B3, which is in line with the 1419 FPS listed in the Model-Zoo [performance chart](https://github.com/Xilinx/Vitis-AI/blob/master/models/AI-Model-Zoo/README.md#performance-on-vck190) for the VCK190. The performance gains moving from C32 to C64 are only about 30% for B1-B4, and 21% for B5. The table below shows the measured FPS for all configurations.

|    |  C32 |  C64 |  speedup |
|:--:|:----:|:----:|:--------:|
| B1 |  482 |  627 |    30.1% |
| B2 |  962 | 1244 |    29.3% |
| B3 | 1436 | 1861 |    29.6% |
| B4 | 1912 | 2477 |    29.6% |
| B5 | 2241 | 2717 |    21.2% |
| B6 | 2846 |    - |        - |

Meanwhile [PG389 Table 2](https://www.xilinx.com/content/dam/xilinx/support/documentation/ip_documentation/dpucvdx8g/v1_1/pg389-dpucvdx8g.pdf#page=11) shows that the theoretical performance is doubled when using C64 instead of C32. I know that these numbers don't necessarily translate to the real world, but the difference seems too big to me.

Is this behavior expected? What could be the bottleneck in the system?

JasonXijieJia, 2022-02-10:
Hi @ksstms 
 Thanks for your kindly feedback!
 Your test results are expected. The doubled theoretical performance from C64 is at the channel dimension. So those small-channel layers will not be benefit from the C64 design, and this will drop the total efficiency.

JasonXijieJia, 2022-02-10:
Hi @ksstms 
  Yes, the performance gain is model-dependent on the DPUCVDX8G design on vck190 board.
  The varies of DPU designs focus on different goals. As introduced in the figure 3 in page 8 of [UG1414](https://www.xilinx.com/content/dam/xilinx/support/documentation/sw_manuals/vitis_ai/2_0/ug1414-vitis-ai.pdf ) , the "G" in the name of DPUCVDX8G refers to "General Propose", and the "H" in the name of DPUCVDX8H refers to "High Throughput"

ksstms, 2022-02-15:
Thank you for the explanation.
I see that the DPUCVDX8H's XO files are available in this repo. Is it possible to implement them on the VCK190? I couldn't find the `arch.json` files for them, so I guess I wouldn't be able to build custom models for those. Are these available to VCK5000 customers only?

JasonXijieJia, 2022-02-16:
Hi @ksstms 
  The DPUCVDX8G's IP and DPUCVDX8H's XO series files are all designed for vc1902 chip, which means for all the boards implementing the chip, including VCK190 and VCK5000. so the different files are the platform and the vitis scripts. You could generate your custom platform on your customized pcb board. 
  DPUCVDX8H's XO series' configurations are fixed. so there is no arch.json file for them to reconfigure, 
  To build custom models for DPUCVDX8H on VCK5000, you can try with their related target name listed in [targets](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/target_factory/targets ), eg [DPUCVDX8H_ISA1_F2W2_8PE](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/target_factory/targets/DPUCVDX8H_ISA1_F2W2_8PE.prototxt) or [DPUCVDX8H_ISA1_F2W4_6PE_aieDWC](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/target_factory/targets/DPUCVDX8H_ISA1_F2W4_6PE_aieDWC.prototxt)