Xilinx/Vitis-AI #674: YOLOX Quantization Error

saz98741, 2022-02-10:
[image]

I get this error, when I quantize the YOLOX model.
I need to deploy YOLOX model on ZCU104, and I use custom dataset for training. Please help me !!

niuxjxlnx, 2022-02-21:
Hi, @saz98741:

The error message is given by pytorch, and it means the model cannot be traced successfully, cannot be deployed either. 
Please change the model input according to the error message, and make the changed model can pass torch.jit.trace() first.

Also you can take YOLO models in Vitis-AI model zoo as reference.

saz98741, 2022-02-25:
So, can I ask about what YOLOX model can deploy on ZCU104 platform?
I use "YOLOX-Darknet53" model training.

Treemann, 2022-02-25:
Hi @saz98741 
YOLOX is released in the latest Vitis-AI model zoo: [pt_yolox_TT100K_640_640_73G_2.0](https://github.com/Xilinx/Vitis-AI/tree/master/models/AI-Model-Zoo/model-list/pt_yolox_TT100K_640_640_73G_2.0)
YOLOX-m is released in Vitis-AI model zoo with some unsupported layers (like Focus, SiLU) replaced and the code is modified for the toolchain & deployment.
Actually you can take the released YOLOX-m as reference and modify your "YOLOX-Darknet53" model, I think it can be deployed if there is not special layer in addition to the original darknet53 structure.

closed 2022-03-09 by niuxjxlnx