Xilinx/Vitis-AI #675: Yolov5 deploy error: Bad any_cast

qw85639229, 2022-02-11:
I'm trying to deploy the yolov5 on ZCU102 with pytorch quantizer.

PeterQuinn396, 2022-02-12:
I'm also experiencing this issue. I successfully built an xmodel using the vitis-ai docker workflow for pytorch, and am now trying to run it on a kv260 ubuntu with vitis-ai runtime 1.3 python API. 

I receive this error when trying to deserialize the xmodel, where `path` is a string with the path location of the xmodel:
```python
File "app.py", line 51, in load_model
    g = xir.Graph.deserialize(path)
RuntimeError: bad any_cast
```

Any insight here would be very appreciated. Thank you

qw85639229, 2022-02-13:
> I'm also experiencing this issue. I successfully built an xmodel using the vitis-ai docker workflow for pytorch, and am now trying to run it on a kv260 ubuntu with vitis-ai runtime 1.3 python API.
> 
> I receive this error when trying to deserialize the xmodel, where `path` is a string with the path location of the xmodel:
> 
> ```python
> File "app.py", line 51, in load_model
>     g = xir.Graph.deserialize(path)
> RuntimeError: bad any_cast
> ```
> 
> I'm attempting to follow the workflow outlined in this example: https://github.com/Xilinx/Vitis-AI-Tutorials/blob/2d611b5842462d2db33a0e92a3858d0e01b6d5ad/Design_Tutorials/09-mnist_pyt/files/application/app_mt.py#L113
> 
> Any insight here would be very appreciated. Thank you

I solved the problem with the Caffe quantizer workflow. By converting Pytorch -> ONNX -> Caffe, the model has successfully run on ZCU102. It seems there is some bug with the Pytorch version.

qianglin-xlnx, 2022-02-22:
HI @qw85639229 
It seems that there is a mismatch of XIR between the Xcompiler in the docker and the VART on the target.
If you use the VAI2.0 docker to compile the model, please also upgrade the VART to VAI2.0 on the target. Thank you.

closed 2022-03-14 by niuxjxlnx

krishnaadithya, 2022-03-31:
@qianglin-xlnx  @qw85639229 

Can you tell me how to deploy yolov5 in xilinx (in my case kv260)

qianglin-xlnx, 2022-04-01:
hi @krishnaadithya 
Yolov5 is supported by Vitis-AI, but the relevant code and model is not published. You can ask the local FAE for them. Thank you.

alpereenkaratas, 2022-10-09:
Hey , @qw85639229. I'm trying to do exactly what you did (quantizing the yolov5, compiling and deploying to the kv260 board). But I have no idea how to write the quantize.py code or how to proceed. I haven't even done the quantization part yet. Can you please share your codes and guide me?

nhatanh174, 2022-12-13:
> I'm trying to deploy the yolov5 on ZCU102 with Pytorch quantizer. I have successfully passed the process of calib, deployment, and compilation. When I try to deserialize the model on ZCU102, this error occurred: [image] there is the previous procedure: Calib: [image]
> 
> Deployment: [image]
> 
> Compilation: [image]

Hi @qw85639229 ,
I am trying to deploy Yolov5 to Xillinx board.
Can you tell me how to quantize and compile it? I
Thanks so much.

IkrameBeggar, 2023-04-25:
> I'm also experiencing this issue. I successfully built an xmodel using the vitis-ai docker workflow for pytorch, and am now trying to run it on a kv260 ubuntu with vitis-ai runtime 1.3 python API. [...]

Hello @PeterQuinn396, I want to quantize a custom yolov5 model but I couldn't find any resource on how to fill the quant.py file. Can you help with that?

qw85639229, 2023-04-25:
> 

It's been a long time. We have finally fixed the problem that you should use the corresponding version of xmodel builder and the vitis-ai runtime.

dungng27, 2023-05-05:
> 

Hi @qw85639229 I wanna ask how do you check the version of xmodel builder and the vitis-ai runtime? I struggle with this problem for long and it'd be a great help if you answer me! Thanks.

feng040107, 2023-05-25:
> I'm trying to deploy the yolov5 on ZCU102 with Pytorch quantizer. I have successfully passed the process of calib, deployment, and compilation. When I try to deserialize the model on ZCU102, this [...]

> I'm trying to deploy the yolov5 on ZCU102 with Pytorch quantizer. I have successfully passed the process of calib, deployment, and compilation. When I try to deserialize the model on ZCU102, this [...]

feng040107, 2023-05-25:
I'm trying to do exactly what you did (quantizing the yolov5, compiling and deploying to the zcu102  board). But I have no idea how to write the quantize.py code or how to proceed. I haven't even done the quantization part yet. Can you please share your codes and guide me?

Muffinxz, 2023-07-29:
@feng040107 Hey pal ! Did you get a response for your issue ?

gaoxinghua951, 2024-01-01:
请问您是怎么修复版本问题的

feng040107, 2024-01-03:
＠qw85639229 請問量化是要修改哪些檔案