Xilinx/Vitis-AI #677: Unable to quantize GlobalAveragePooling2D due to missing attribute

JurgenVandendriessche, 2022-02-17:
Hello,

When I quantize a model with a global average pooling layer, I get the following error.

>   File "quantize.py", line 67, in <module>
    main()
  File "quantize.py", line 64, in main
    quantize(conf)
  File "quantize.py", line 23, in quantize
    quantized_model = quantizer.quantize_model(calib_dataset=dataset)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 578, in quantize_model
    calib_steps)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 271, in _calibrate_without_loss
    mode='QCB')
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 745, in create_quantize_model
    quantize_strategy, mode)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 802, in quantize_apply
    quantize_strategy.get_quantize_registry(), mode)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 168, in apply
    layer_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 50, in _apply_availables
    layer_metadata).recursive_transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 738, in recursive_transform
    self.layer_metadata).transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 696, in transform
    transformed_model = keras.Model.from_config(self._config, custom_objects)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/training.py", line 2397, in from_config
    functional.reconstruct_from_config(config, custom_objects))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/functional.py", line 1283, in reconstruct_from_config
    process_node(layer, node_data)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/functional.py", line 1231, in process_node
    output_tensors = layer(input_tensors, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 977, in __call__
    input_list)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 1115, in _functional_construction_call
    inputs, input_masks, args, kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 848, in _keras_tensor_symbolic_call
    return self._infer_output_signature(inputs, args, kwargs, input_masks)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 886, in _infer_output_signature
    self._maybe_build(inputs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 2659, in _maybe_build
    self.build(input_shapes)  # pylint:disable=not-callable
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/layers/vitis_pooling.py", line 104, in build
    if self._is_global_pooling(input_shape):
AttributeError: 'VitisGlobalAveragePooling2D' object has no attribute '_is_global_pooling'

When I look at the source code of [vitis_pooling.py](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_tensorflow2.x/tensorflow_model_optimization/python/core/quantization/keras/vitis/layers/vitis_pooling.py), it seems that the function '_is_global_pooling' is only implemented for the normal average pooling layer and missing for the global average pooling layer.

chuo1, 2022-07-21:
> Hello,
> 
> When I quantize a tensorflow 2 model with a global average pooling layer, I get the following error.
> 
> > File "quantize.py", line 67, in 
> > main()
> > File "quantize.py", line 64, in main
> > quantize(conf)
> > File "quantize.py", line 23, in quantize
> > quantized_model = quantizer.quantize_model(calib_dataset=dataset)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 578, in quantize_model
> > calib_steps)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 271, in _calibrate_without_loss
> > mode='QCB')
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 745, in create_quantize_model
> > quantize_strategy, mode)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 802, in quantize_apply
> > quantize_strategy.get_quantize_registry(), mode)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 168, in apply
> > layer_metadata)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 50, in _apply_availables
> > layer_metadata).recursive_transform()
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 738, in recursive_transform
> > self.layer_metadata).transform()
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 696, in transform
> > transformed_model = keras.Model.from_config(self._config, custom_objects)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/training.py", line 2397, in from_config
> > functional.reconstruct_from_config(config, custom_objects))
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/functional.py", line 1283, in reconstruct_from_config
> > process_node(layer, node_data)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/functional.py", line 1231, in process_node
> > output_tensors = layer(input_tensors, **kwargs)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 977, in **call**
> > input_list)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 1115, in _functional_construction_call
> > inputs, input_masks, args, kwargs)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 848, in _keras_tensor_symbolic_call
> > return self._infer_output_signature(inputs, args, kwargs, input_masks)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 886, in _infer_output_signature
> > self._maybe_build(inputs)
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/engine/base_layer.py", line 2659, in _maybe_build
> > self.build(input_shapes)  # pylint:disable=not-callable
> > File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/layers/vitis_pooling.py", line 104, in build
> > if self._is_global_pooling(input_shape):
> > AttributeError: 'VitisGlobalAveragePooling2D' object has no attribute '_is_global_pooling'
> 
> When I look at the source code of [vitis_pooling.py](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_tensorflow2.x/tensorflow_model_optimization/python/core/quantization/keras/vitis/layers/vitis_pooling.py), it seems that the function '_is_global_pooling' is only implemented for the normal average pooling layer and missing for the global average pooling layer.

hi, i try to open the source code of [vitis_pooling.py](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_tensorflow2.x/tensorflow_model_optimization/python/core/quantization/keras/vitis/layers/vitis_pooling.py), but it shows 404 not found. In addition, vai2.5 version has been released, which contains the global average pooling layer, you can try it.

JurgenVandendriessche, 2022-07-26:
Hello,

Indeed, the url is no longer valid.
[This](https://github.com/Xilinx/Vitis-AI/blob/2.0/tools/Vitis-AI-Quantizer/vai_q_tensorflow2.x/tensorflow_model_optimization/python/core/quantization/keras/vitis/layers/vitis_pooling.py) is an updated link to branch 2.0.
It also looks like the bug is fixed in Vitis AI 2.5 so I will close this issue.

closed 2022-07-26 by JurgenVandendriessche