Xilinx/Vitis-AI #679: Why "vart" module is missing in pytorch environment?

vittofa, 2022-02-21:
Hello,

I downloaded and builded Vitis-AI 2.0.0.1103 for GPU, when I activate pytorch environment and I try to load "vart" module in python script I get the following error:
[image]
What could be the problem?

Before Vitis-AI 2.0 I was using version 1.4 and the module "vart" was imported without any problem in the pytorch environment.

Thanks.

qianglin-xlnx, 2022-02-23:
Hi @vittofa 
Thank you for your feedback.
There is dependency issue with integration of pytorch 1.7, vart, rt-engine in pytorch conda env.
I have file a jira internal to solve this issue.
Before we solve it, could you load vart model in the base conda environment? Thank you.

qianglin-xlnx, 2022-07-05:
Hi @vittofa 
Vai2.5 is released. If possible, you can test it with the VAI2.5 docker. Thank you.

qianglin-xlnx, 2022-08-31:
Closing since no activity for more than 1 month, please reopen if you still have question, thanks

closed 2022-08-31 by qianglin-xlnx