Xilinx/Vitis-AI #682: torch_quantizer results in RuntimeError: CUDA out of memory.

910560, 2022-02-22:
Would like to know how to reduce the memory allocated by pytorch 1.7?

niuxjxlnx, 2022-02-22:
Hi, @910560:

Are you using Vitis-AI 2.0? 
Can your float model run on GPU? and how much GPU memory is needed?
How much GPU memory is available on your server?

910560, 2022-02-23:
Vitis-AI 2.0, GPU, GeForce GTX 1080 Ti, 11GB

Here is the error message:

RuntimeError: CUDA out of memory. Tried to allocate 1.88 GiB (GPU 0; 10.92 GiB total capacity; 9.42 GiB already allocated; 853.44 MiB free; 9.43 GiB reserved in total by PyTorch)

910560, 2022-02-23:
I did google for answers
Some said that can use [torch.cuda.set_per_process_memory_fraction(fraction, device=None)] to limit PyTorch GPU memory allocation.
But it is not available in PyTorch 1.7.1 which is bundled in the Vitis-AI docker image.

niuxjxlnx, 2022-02-23:
Can your float model run on GPU by pytorch?

910560, 2022-02-23:
Yes, it can

niuxjxlnx, 2022-02-23:
torch_quantizer does need more GPU memory than pytorch because more tensors needs help store quantization information.

910560, 2022-02-23:
is there anyway i can upgrade the pytorch version while keeping the torch_quantizer and relative package runs correctly?

niuxjxlnx, 2022-02-23:
You can refer to the part of script replace_pytorch.sh in user guide. And get the updated script at https://github.com/Xilinx/Vitis-AI/issues/666.

closed 2022-02-28 by niuxjxlnx