Xilinx/Vitis-AI #683: Compile problem with U250 and vai_c_xir

vittofa, 2022-02-22:
Hello,
I'm using Vitis-AI 2.0.0.1103, I need to deploy my Neural Network for Alveo U250.
After the release of Vitis-AI 2.0 the compilation of the model has been possible but I'm facing some issue with the deployment, in particular when I compile the model for the target architecture (DPUCADF8H_ISA0), I then run the "xdputil benchmark" command I get the following error:
[image]
To compile the xmodule I used this command:
[image]
To be shure that "xdputil benchmarck" is working correctly I've downloaded xilinx pre-compiled xmodel for U250 in particular "inception_v1_tf.xmodel" and the benchmark is executed without any porblem, so searching for difference between the models I came across that batch size is different... If I run "xdputil xmodel <name.xmodel> -l" with xilinx's inception model I get:
[image]
But when I run the same command for my xmodel I get:
[image]
Even thought I added the input shape option in the compiling command...
What's going on? I don't understand how to fix it.

Thank you.
Vittorio

qianglin-xlnx, 2022-02-24:
HI @vittofa 
I confirm from my colleague that for pytorch model, nndct tool only support batch number=1.

SuZhan0322, 2022-09-01:
I think we can close this issue now. Hi @vittofa, please reopen this issue if there is any problem.