Xilinx/Vitis-AI #684: Reid Quantization Error by pytorch

jzymessi, 2022-02-23:
I used JDAI-CV/fast-reid(https://github.com/JDAI-CV/fast-reid) trained a  model(backbone  resnest ) , when I use  pytorch  quantize the model , The following error occurred :
[image]
Does anyone know why?  Thanks.

Alexiazzf, 2022-03-10:
Hi @jzymessi ,

It seems like there is something wrong with conv channels. Is there any error when you run float model? Can you upload  your torch model or show more details?

qianglin-xlnx, 2022-06-11:
Hi @jzymessi 
Has the problem been solved?

jzymessi, 2022-06-16:
> Hi @jzymessi Has the problem been solved?

Hi @qianglin-xlnx @Alexiazzf 
This problem has been solved,Thanks!

closed 2022-06-16 by jzymessi