Xilinx/Vitis-AI #685: An error of the process of converting to xmodel with Vitis_ai_pytorch

fathersdog, 2022-02-24:
hello:
when I run the code: python -u quantize.py -d ${BUILD} -b 1 --quant_mode test 2>&1 | tee ${LOG}/quant_test.log
I will get:
`
[VAIQ_NOTE]: =>Converting to xmodel ...
Traceback (most recent call last):
  File "quantize.py", line 110, in <module>
    run_main()
  File "quantize.py", line 104, in run_main
    quantize(args.build_dir, args.quant_mode, args.batchsize)
  File "quantize.py", line 79, in quantize
    quantizer.export_xmodel(deploy_check=False, output_dir=quant_model)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 110, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 180, in export_xmodel
    dump_xmodel(output_dir, deploy_check, self._lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 203, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 297, in get_deploy_graph_list
    deploy_graphs = g_optmizer.partition_by_quant_part()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/deploy_optimizer.py", line 581, in partition_by_quant_part
    partition_check(quant_dev_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/deploy_optimizer.py", line 560, in partition_check
    raise RuntimeError(f"Please check graph partition: the quant node '{node.name}' should be in quant graph.")
RuntimeError: Please check graph partition: the quant node 'CLReg::CLReg/Exp2_Decoder26[decoder]/4902' should be in quant graph.
`

niuxjxlnx, 2022-03-01:
Hi, @fathersdog:

In general, this error is because of wrong partial quantization setting by QuantStub/DequantStub or custom OP. You can upload your model script here if this hint cannot help you solve the problem directly.

closed 2022-03-11 by niuxjxlnx

fathersdog, 2022-03-12:
Thank you for your reply, I have solved the problem and found that the quantization of Vitis-ai does not support the torch.split function.