Xilinx/Vitis-AI #686: Unable to map APM params When using vaitrace

Richsheep, 2022-02-25:
Hi,

I met an error when using vaitrace to profile DPU performance on my own board.

this error cannot reproduce on zcu104 board.

Detail:
```
# vaitrace -d --xat ./dpu_inference
DEBUG:root:Cmd line args:
DEBUG:root:Namespace(bypass=False, cmd=['./dpu_inference'], config='', debug=True, python=False, showversion=False, timeout=None, traceSaveTo=None, va=False, xat=True)
DEBUG:root:Use boot as trace clock
INFO:root:VART will run xmodel in [NORMAL] mode
DEBUG:root:Xmodel not exist: []
DEBUG:root:Control Options:
DEBUG:root:{'cmd': ['./dpu_inference'], 'xat': {'compress': True, 'filename': '/home/root/dpu_inference/dpu_inference.xat'}, 'config': None, 'timeout': 5, 'debug': True, 'traceClock': 'boot', 'x86_tsc_khz': 0, 'platform': {'release': ['5', '4'], 'machine': 'aarch64'}, 'runmode': 'normal'}
DEBUG:root:Preparing [xrt]...
DEBUG:root:Compatible [xrt]: True
DEBUG:root:Preparing [hwInfo]...
DEBUG:root:Compatible [hwInfo]: True
DEBUG:root:Preparing [function]...
DEBUG:root:Compatible [function]: True
Analyzing symbol tables...
61 / 61
51 / 51
3 / 3
32 / 32
DEBUG:root:Preparing [sched]...
DEBUG:root:Compatible [sched]: True
DEBUG:root:Preparing [xapm]...
DEBUG:root:Compatible [xapm]: True
Unable to map APM params
DEBUG:root:Preparing [cmd]...
DEBUG:root:Compatible [cmd]: True
DEBUG:root:Preparing [DPUController]...
DEBUG:root:Compatible [DPUController]: True
DEBUG:root:Preparing [pyfunc]...
DEBUG:root:Compatible [pyfunc]: True
Segmentation fault
```

Any Suggestions to Debug this case?

Many Thanks.

li900309, 2022-02-25:
Hi, 
I think, it requires AMP driver, please add the xapm driver to your linux kernel

1. The following config option of linux kernel should be enabled in order to build the apm driver:
config UIO_XILINX_APM
2. The following device tree node should be added to dts:
apm: apm@FD0B0000 {
    xlnx,enable-profile = <0>;
    xlnx,enable-trace = <0>;
    xlnx,num-monitor-slots = <4>;
    xlnx,enable-event-count = <1>;
    xlnx,enable-event-log = <1>;
    xlnx,have-sampled-metric-cnt = <1>;
    xlnx,num-of-counters = <8>;
    xlnx,metric-count-width = <32>;
    xlnx,metrics-sample-count-width = <32>;
    xlnx,global-count-width = <32>;
    xlnx,metric-count-scale = <1>;
    xlnx,id-filter-32bit;
    status = "okay";
    compatible = "xlnx,axi-perf-monitor";
    interrupt-parent = <&&gic>;                /* Check for Interrupt Controller Node
    clocks = <&&clkc 28>;                      /* Check for the CLK */
    interrupts = <0x0 123 0x4>;                /* Check for the interrupt number */
    reg = <0x0 0xFD0B0000 0x0 0x10000>;
};

the official document:
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842046/APM

Richsheep, 2022-03-01:
HiThank

Richsheep, 2022-03-01:
Hi,

Thank you for your support

According to your reply, I realize I need add UIO_XILINX_APM to Kernel config.

Should I inplement an axi_perf_mon IP in Block Design and add apm defination in device tree?

or  I can only add apm Kernel config for my usage (if I only want to know DPU performance with vaitrace)

li900309, 2022-03-01:
Hi, no need to add a new axi_perf_mon in block design,There are 4 APMs available on the Zynq Ultrascale PS side, vaitrace can  can use it, just enable the driver in kernel config and add node  **apm: apm@FD0B0000 {....}** (the same as show in the wiki page "PS APM DDR Example")  to your device tree, then it can work

Richsheep, 2022-03-01:
> Hi, no need to add a new axi_perf_mon in block design,There are 4 APMs available on the Zynq Ultrascale PS side, vaitrace can can use it, just enable the driver in kernel config and add node **apm: [...]

Many thanks!  I'll try it.

Richsheep, 2022-03-08:
> Hi, no need to add a new axi_perf_mon in block design,There are 4 APMs available on the Zynq Ultrascale PS side, vaitrace can can use it, just enable the driver in kernel config and add node **apm: [...]

Thank you again, finally I got the correct result after follow your guidence.

closed 2022-03-08 by Richsheep