Xilinx/Vitis-AI #687: What exactly does the “Gops in the name of each model mean and how did it get there?

Ru-Huang, 2022-02-28:


Ru-Huang, 2022-02-28:
Hi，I've been doing some testing with your model recently. The whole tutorial is well done. But there is one place I don't understand, what exactly does the "Gops" mean and how you get this value ? 
[image]

and can you explain  what is the difference between total trainable params and "Gops" here?
[image]

Ru-Huang, 2022-02-28:
I'm just getting started with neural networks, and there are a lot of things I don't understand.  hope you can help me！

wangxd-xlnx, 2022-03-01:
Hi @Serendi-pity1995 

It's actually a shorthand for G FLOPs. The FLOPs (Floating Point Operations) is used to measure the computational complexity of a model, and is often used as an indirect measure of the speed of a model.

So can sum up, it represents the amount of calculation. And ‘params’ is a completely different concept.

Serendi-pity1995, 2022-03-01:
> 

Thanks for your reply, but how do I get the Gops of each model? I saw someone write that the xir tool can be used, is that right? Does the content in subg_root in the generated txt file represent Gops?

[image]
I used the tool xir to generate the info of the model "tf2_efficientnet-b0_imagenet_224_224_0.36G_2.0" but the numbers here do not match it (0.36G).
I also tried to get infos of some other models.  In "tf_ssdinceptionv2_coco_300_300_9.62G_2.0" ,the number is 9,625,770,531
In "tf_mobilenetv2_1.0_imagenet_224_224_602M_2.0" ,the number is 608,725,705 . It looks like it fits right in.
It's just a coincidence, or I've got the wrong numbers.

Serendi-pity1995, 2022-03-02:
> Hi @Serendi-pity1995
> 
> There‘re many open source tools to calculate flops. We recommend 'ptflops' or 'thop'. But please note that 1 MACs = 2 Flops.
> 
> Such as thop, please refer to https://github.com/Lyken17/pytorch-OpCounter.

Many thanks！

closed 2022-03-02 by Serendi-pity1995