Xilinx/Vitis-AI #688: AboutVitis AI support

yumianhuli2, 2022-03-02:
1. Does Vitis AI currently support FPGA ArtiX-7 Series?

2. Could you plz list all the FPGA series currently supported by Vitis AI?
Thank you!

sudharsan2000, 2022-04-14:
@andyluo7 @qianglin-xlnx any help would greatly help.
Is there any flow for using the images such as this one for [ZCU104](https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-zcu104-dpu-v2021.2-v2.0.0.img.gz) mentioned under this [page](https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART).

li900309, 2022-04-14:
Q3: as far as I know, DPU can not support QEMU emulator

closed 2022-06-11 by qianglin-xlnx

SteCla0, 2022-06-22:
Same question.
I don't understand why it is closed without reply/answer