Xilinx/Vitis-AI #690: import caffe in python does not work

mhanuel26, 2022-03-02:
I have the issue that I cannot do the following import under the python folder in caffe_xilinx folder

```
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/caffe-xilinx > python
Python 3.6.15 | packaged by conda-forge | (default, Dec  3 2021, 18:49:41) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from caffe.proto import caffe_pb2
>>> 
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/caffe-xilinx > cd python/
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/caffe-xilinx/python > python
Python 3.6.15 | packaged by conda-forge | (default, Dec  3 2021, 18:49:41) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from caffe.proto import caffe_pb2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/workspace/models/AI-Model-Zoo/caffe-xilinx/python/caffe/__init__.py", line 1, in <module>
    from .pycaffe import Net, SGDSolver, NesterovSolver, AdaGradSolver, RMSPropSolver, AdaDeltaSolver, AdamSolver
  File "/workspace/models/AI-Model-Zoo/caffe-xilinx/python/caffe/pycaffe.py", line 13, in <module>
    from ._caffe import Net, SGDSolver, NesterovSolver, AdaGradSolver, \
ModuleNotFoundError: No module named 'caffe._caffe'
>>> 
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/caffe-xilinx/python > 
```

Before on another station I believe I solved the issue by adding a soft link inside caffe-xilinx folder to the actual bin where the imports are located, for instance I did 

`sudo ln -s /opt/vitis-ai/conda/envs/vitis-ai-caffee/bin /workspace/models/AI-Model-Zoo/caffe-xilinx`

but that is not working on my new environment.

I have check the path as 

```
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/caffe-xilinx/python > echo $PATH
/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin:/opt/vitis_ai/conda/bin:/opt/vitis_ai/conda/condabin:/usr/local/nvidia/bin:/usr/local/cuda/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
```

The reason I am looking into this is because I am running the data preparation step of cf_refinedet_coco_360_480_0.96_5.08G_2.0, the bash call create_data.py which indeed calls the create_annoset.py under the caffe-xilinx/python folder, line 90 of the file looks like this

`caffe_py_path = caffe_root + 'python/create_annoset.py`

Does anyone have any idea how to fix this?

Thanks in advance,

wangxd-xlnx, 2022-03-11:
Hi @mhanuel26 

We have analyzed this issue with #691 #694 and we think these are the same kind of problem.

Please notice that if you could use Vitis-AI docker, you just need to activate the conda env ‘vitis-ai-caffe'. You don't need to compile the caffe-xilinx source code manually. It's just for the environment that is inconvenient to use docker.

The caffe-xilinx is precompiled in conda env ‘vitis-ai-caffe'. You can directly use it. 

PATH: /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/

[image]

qianglin-xlnx, 2022-04-07:
Hi @mhanuel26 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-04-07 by qianglin-xlnx