Xilinx/Vitis-AI #692: Vitis AI 2.0 Custom OP Plug-in: Develop Custom-Op with MaxPool3d and AvgPool3d for PyTorch

MarvS95, 2022-03-03:
Hello,

I am trying to quantize and compile a PyTorch model with two operations (MaxPool3d and AvgPool3d) which are not supported by Vitis AI 2.0 yet. Therefore I want to develop two custom operations as described in the User Guide:

https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Pytorch-Custom-OP-Model-Example 

So far there is one example how to develop a custom op for a PyTorch model but I don't know how to apply this workflow to my model. The following code snippet is from my model definition and shows the usage of MaxPool3d, which is not supported. Does anyone here have some experience with custom op plug-in and can help me?

```
        self.pool_layer = nn.MaxPool3d(
            kernel_size=[1, 3, 3], stride=[1, 2, 2], padding=[0, 1, 1]
        )

    def forward(self, x):
        x = self.conv(x)
        x = self.bn(x)
        x = self.relu(x)
        x = self.pool_layer(x)
        return x
```

Does anyone here have some experience with custom op plug-in and can help me?

Many thanks

niuxjxlnx, 2022-03-04:
Hi, @MarvS95:

This example model should easier for your reference: https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant_custom_op.py.
You can refer to the function "custom_op".

MarvS95, 2022-03-04:
Hi @niuxjxlnx

thank you really much for this example. It's much easier for me to understand. 

I moved the usage of the MaxPool3d into the definition of my custom op and decorated it as follows:

```
@register_custom_op(op_type="MaxPool3d_Custom_Op", attrs_list=[])
def MaxPool3d_Custom_Op(ctx):
    pool = nn.MaxPool3d(
        kernel_size=[1, 1, 1],
        stride=[1, 1, 1],
        padding=[0, 0, 0],
    )
    return pool
```

Then I changed it's calling accordingly in the model definition:

```
class SlowFastModel(nn.Module):

    pool = MaxPool3d_Custom_Op() # this is line 193 from error message 

    def forward(self, x):
        x = pool(x)
        return x
```

When i  run the quantizer-script, I get the following error:

```
  File "/workspace/SlowFast_Custom_Op_Quantizer/video_model_builder.py", line 193, in _construct_network
    pool = MaxPool3d_Custom_Op()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/utils/op_register.py", line 101, in innner
    return custom_op.apply(*args, **kwargs)
TypeError: MaxPool3d_Custom_OpBackward.forward: expected Tensor or tuple of Tensor (got MaxPool3d) for return value 0
```

Do you know how I can handle the tensor input to fix this problem? 

Thank you for your help

niuxjxlnx, 2022-03-07:
Hi, @MarvS95:

The return value of function "MaxPool3d_Custom_Op" should be the calculation result of maxpool3d, not a maxpool3d OP. You  should write the function like the following:

@register_custom_op(op_type="MaxPool3d_Custom_Op", attrs_list=[...])
def MaxPool3d_Custom_Op(ctx, input, ...):
    return torch.nn.functional.max_pool3d(input, ...)

MarvS95, 2022-03-07:
Hi @niuxjxlnx 

thank you, now I understand what the problem was. The layer definition needs the class MaxPool3d and forward() needs the result of the calculation. I copied the class MaxPool3d and the necessary imports into a python-file where I register the method forward() of the class MaxPool3d as a custom op:

```
class MaxPool3d(_MaxPoolNd):

    kernel_size: _size_3_t
    stride: _size_3_t
    padding: _size_3_t
    dilation: _size_3_t

    @register_custom_op(op_type="Custom_Op_forward", attrs_list=["input"])
    def forward(self, input: torch.Tensor) -> torch.Tensor:
        return F.max_pool3d(input, self.kernel_size, self.stride,
                            self.padding, self.dilation, self.ceil_mode,
                            self.return_indices)
```
The next one is the layer definition. It defines two MaxPool3d operations, one for each pathway of the model. I replaced nn.MaxPool3d by MaxPool3d. The class is the same but now it comes from the python-file where I register the custom op:

```
      (...)    

      for pathway in range(self.num_pathways):
            pool = MaxPool3d(
                kernel_size=pool_size[pathway],
                stride=pool_size[pathway],
                padding=[0, 0, 0],
            )
            self.add_module("pathway{}_pool".format(pathway), pool)

      (...)
```
And here is the call of pool in method forward() of the model (for each pathway):

```
    def forward(self, x):
        (...)
        for pathway in range(self.num_pathways):
            pool = getattr(self, "pathway{}_pool".format(pathway))
            x[pathway] = pool(x[pathway])
        return x
        (...)
```
Now I get this error message:
```
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/utils/op_register.py", line 101, in innner
    return custom_op.apply(*args, **kwargs)
TypeError: forward() takes 2 positional arguments but 3 were given
```
I don't understand which 2 positional arguments are meant here and I also don't understand why 3 were given. When I remove the line @register_custom_op(...), I get no errors. Do you have an idea what's wrong?

MarvS95, 2022-03-07:
Hi @niuxjxlnx 

thank you for this little example. Now the not supported functions seem to be registered as custom ops. Strangely, when building the .xmodel, the quantization process stops with a warning that one of these operations is not defined in XIR. There is also an error message about an unsupported data type before aborting:

```
(vitis-ai-pytorch) Vitis-AI /workspace/SlowFast_Custom_Op_Quantizer > sh run_all.sh 
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/cuda/__init__.py:52: UserWarning: CUDA initialization: Found no NVIDIA driver on your system. Please check that you have an NVIDIA GPU and installed a driver from http://www.nvidia.com/Download/index.aspx (Triggered internally at  /opt/conda/conda-bld/pytorch_1607370120218/work/c10/cuda/CUDAFunctions.cpp:100.)
  return torch._C._cuda_getDeviceCount() > 0
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[VAIQ_NOTE]: Loading NNDCT kernels...

[VAIQ_NOTE]: `AvgPool3d_Custom_Op_Head_pathway0` has been registered as a custom op.

[VAIQ_NOTE]: `AvgPool3d_Custom_Op_Head_pathway1` has been registered as a custom op.

[VAIQ_NOTE]: `MaxPool3d_Custom_Op_Stem_pathway0` has been registered as a custom op.

[VAIQ_NOTE]: `MaxPool3d_Custom_Op_Stem_pathway1` has been registered as a custom op.

[VAIQ_NOTE]: `MaxPool3d_Custom_Op` has been registered as a custom op.
Running on CPU!

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing SlowFastModel...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 371/371 [00:00<00:00, 920.70it/s, OpInfo: name = SlowFastModel/ResNetBasicHead[head]/22326, type = mean]                                                                         

[VAIQ_WARN]: The quantizer recognize new op `AvgPool3d_Custom_Op_Head_pathway1` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `MaxPool3d_Custom_Op_Stem_pathway0` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `MaxPool3d_Custom_Op` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `MaxPool3d_Custom_Op_Stem_pathway1` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `AvgPool3d_Custom_Op_Head_pathway0` as a float operator by default.

[VAIQ_NOTE]: =>Doing weights equalization...
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/optimization/commander.py:484: RuntimeWarning: divide by zero encountered in true_divide
  scale = np.where(sqrt_of_ranges != 0, range_0 / sqrt_of_ranges, scale)

[VAIQ_NOTE]: =>Quantizable module is generated.(build/quant_model/SlowFastModel.py)

[VAIQ_NOTE]: =>Get module with quantization.
Quantizing Model: 100%|███████████████████████████████████████████████| 1/1 [00:37<00:00, 37.78s/it]

[VAIQ_NOTE]: =>Exporting quant config.(build/quant_model/quant_info.json)

[VAIQ_WARN]: weight SlowFastModel::s4.pathway0_res0.branch2.b.weight value is too small, so adjust the fix position from 10 to 9

[VAIQ_WARN]: weight SlowFastModel::s4.pathway0_res4.branch2.b.weight value is too small, so adjust the fix position from 10 to 9

[VAIQ_WARN]: weight SlowFastModel::s4_fuse.conv_f2s.weight value is too small, so adjust the fix position from 10 to 9

[VAIQ_WARN]: weight SlowFastModel::s5.pathway0_res0.branch1.weight value is too small, so adjust the fix position from 10 to 9

[VAIQ_WARN]: weight SlowFastModel::s5.pathway0_res0.branch2.a.weight value is too small, so adjust the fix position from 10 to 9

[VAIQ_WARN]: weight SlowFastModel::s5.pathway0_res0.branch2.b.weight value is too small, so adjust the fix position from 10 to 9

[VAIQ_WARN]: weight SlowFastModel::s5.pathway0_res1.branch2.a.weight value is too small, so adjust the fix position from 10 to 8
[... 28 lines cut ...]
[VAIQ_WARN]: The quantizer recognize new op `MaxPool3d_Custom_Op_Stem_pathway1` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `AvgPool3d_Custom_Op_Head_pathway0` as a float operator by default.

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(build/quant_model/SlowFastModel.py)

[VAIQ_NOTE]: =>Get module with quantization.
Testing Model: 100%|██████████████████████████████████████████████████| 1/1 [00:21<00:00, 21.61s/it]
Network output shape = torch.Size([400])

[VAIQ_NOTE]: =>Converting to xmodel ...

[VAIQ_NOTE]: =>Dumping 'SlowFastModel'' checking data...

[VAIQ_NOTE]: =>Finsh dumping data.(build/quant_model/deploy_check_data_int/SlowFastModel)
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0307 16:24:46.159703   315 tool_function.cpp:188] [UNILOG][WARNING] The operator named SlowFastModel__SlowFastModel_VideoModelStem_s1__ResNetBasicStem1_pathway1_stem__18889, type: MaxPool3d_Custom_Op_Stem_pathway1, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
F0307 16:24:46.163223   315 wrapper.cpp:111] [UNILOG][FATAL][PYXIR_INVALID_DATA_TYPE][] Unsupported data type!
*** Check failure stack trace: ***
Aborted (core dumped)
```

Are the warnings [VAIQ_WARN] critical? And do you have any idea how I can find out the cause of error F0307?

carrotandjzy, 2022-03-08:
Can you paste your custom ops code ?

MarvS95, 2022-03-09:
Hi @carrotandjzy,

I found an other problem in my custom op usage and made some changes yesterday: I copied the class MaxPool3d and the necessary imports into a python-file where I also register the custom op. I call the custom op in the forward() method of the class MaxPool3d:

```
@register_custom_op(op_type="custom_pool_3d", attrs_list=["kernel_size", "stride",
                            "padding", "dilation", "ceil_mode",
                            "return_indices"])
def custom_pool_3d(ctx, input: torch.Tensor, kernel_size: tuple, stride: tuple,
                            padding: tuple, dilation: tuple, ceil_mode: bool,
                            return_indices: bool) -> torch.Tensor:
    return F.max_pool3d(input, kernel_size, stride,
                        padding, dilation, ceil_mode,
                        return_indices)
 
 
class MaxPool3d(_MaxPoolNd):
 
    kernel_size: _size_3_t
    stride: _size_3_t
    padding: _size_3_t
    dilation: _size_3_t
 
    def forward(self, input: torch.Tensor) -> torch.Tensor:
        return custom_pool_3d(input, self.kernel_size, self.stride, self.padding, self.dilation,
                              self.ceil_mode, self.return_indices)
```
Do you think this is the right approach now to register the custom op?

Now I get an error because of the unknown type 'tuple' (which is the type of the parameters kernel_size, stride, padding and dilation of MaxPool3d):

```
Traceback (most recent call last):
  File "SlowFast_quant.py", line 286, in <module>
    main()
  File "SlowFast_quant.py", line 281, in main
    quantize_model(model, 'calib', device, quant_stub, quant_dataset, args.output_dir, args.batch_size)
  File "SlowFast_quant.py", line 203, in quantize_model
    quantizer = pna.torch_quantizer(quant_mode, model, quant_stub, output_dir=str(output_path), device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 89, in __init__
    custom_quant_ops = custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 126, in __init__
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 196, in prepare_quantizable_module
    graph = parse_module(module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 78, in parse_module
    module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/parser.py", line 68, in __call__
    raw_graph, raw_params = graph_handler.build_torch_graph(graph_name, module, input_args) 
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/trace_helper.py", line 45, in build_torch_graph
    raw_graph, raw_params = self._build_raw_graph(graph_name, fw_graph, params)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/trace_helper.py", line 164, in _build_raw_graph
    torch_value = TorchValue(value, name=f"extra_{extra_count}")
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/torch_graph.py", line 210, in __init__
    raise RuntimeError(f"The type of value ({type(value)}) is unkown.")
RuntimeError: The type of value (<class 'tuple'>) is unkown.
```
Since MaxPool3d can also take single integers for kernel_size, stride, padding and dilation (if same value is used for the depth, height and width dimension), I tried to pass single integers. That fixed the problem:

```
pool = MaxPool3d(
                kernel_size=1,
                stride=1,
                padding=0,
            )
```

But I also have usages with different values for the three dimensions, e.g. kernel_size=(1, 3, 3). Do you have an idea why I can't pass tuple or how I can pass different values without using tuple? (same error with list) 

Thank you very much for your help

carrotandjzy, 2022-03-10:
Hi , @MarvS95 
I have checked the soure code and found that the tuple / list type of attr are not supported in quantizer. You can modify code to walk around this issue.
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/parse/torch_graph.py:
line 198:   elif isinstance(value, (float, int, bool)): =>   **elif isinstance(value, (float, int, bool, tuple, list)):**
line 205:  self._type = {float: 'float', int: 'int',  bool: 'bool'}.get(type(value), None)  => **self._type = str(type(value))**

MarvS95, 2022-03-10:
Hi @carrotandjzy 

thank you very much, that solved this problem! Now (I think) I have only one problem left before I can generate the xmodel:

```
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0310 03:33:15.232961   901 tool_function.cpp:188] [UNILOG][WARNING] The operator named SlowFastModel__SlowFastModel_VideoModelStem_s1__ResNetBasicStem_pathway1_stem__MaxPool3d_pool_layer__19015, type: custom_max_pool_3d, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
W0310 03:33:15.292793   901 tool_function.cpp:188] [UNILOG][WARNING] The operator named SlowFastModel__SlowFastModel_ResNetBasicHead_head__AvgPool3d_pathway1_avgpool__22412, type: custom_avg_pool_3d, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
F0310 03:33:15.367183   901 shape_inference.cpp:1779] [UNILOG][FATAL][XIR_INVALID_ARG_OCCUR][Invalid arg occurence!] xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__Linear_projection__input, type = matmul}'s bias' shape is {1, 2, 400, 2}, but xir only support one dimension bias.
*** Check failure stack trace: ***
Aborted (core dumped)
```

I get the error [UNILOG][FATAL][XIR_INVALID_ARG_OCCUR][Invalid arg occurence!]. As far as I understand it says, that the bias' shape of the op SlowFastModel__SlowFastModel_ResNetBasicHead_head__Linear_projection__input (torch.nn.Linear) has more than one dimension with a shape of (1, 2, 400, 2). 

In the definition of this op I checked the bias' shape as follows:

```
        self.projection = nn.Linear(sum(dim_in), num_classes, bias=True)
        print("BIAS:", format((self.projection.state_dict()['bias']).shape))
```
The output is: BIAS: torch.Size([400])

So the bias seems to be one dimensional with the size of num_classes which is 400. Do you know what causes the error?
When I comment this function, a SlowFastModel_int.xmodel is generated. 

Thank you very much.

carrotandjzy, 2022-03-11:
Can you provide your model  and dummy input ?

MarvS95, 2022-03-14:
Hi @carrotandjzy 

I found a way to walk around the error with the bias' shape by deleting the following line:

https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/modules/linear.py

line 132: self.bias.data = torch.sub(self.bias.data, error, alpha=rate)

Now I get SlowFastModel_int.xmodel but can't compile:

```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][WARNING] The operator named SlowFastModel__SlowFastModel_VideoModelStem_s1__ResNetBasicStem_pathway0_stem__MaxPool3d_pool_layer__18966, type: custom_max_pool_3d, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
[UNILOG][WARNING] The operator named SlowFastModel__SlowFastModel_ResNetBasicHead_head__AvgPool3d_pathway0_avgpool__22395, type: custom_avg_pool_3d, is not defined in XIR. XIR creates the definition of this operator automatically. You should specify the shape and the data_type of the output tensor of this operation by set_attr("shape", std::vector<int>) and set_attr("data_type", std::string)
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: SlowFastModel, with op num: 864
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__22431, type = concat-fix} has been assigned to CPU: [Input xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__AvgPool3d_pathway0_avgpool__22395_inserted_fix_8, type = float2fix} is not in DPU subgraph. And output dimension is not 4.].
[UNILOG][WARNING] xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__22431, type = concat-fix} has been assigned to CPU: [Input xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__AvgPool3d_pathway0_avgpool__22395_inserted_fix_8, type = float2fix} is not in DPU subgraph. And output dimension is not 4.].
Killed
```

How can I send the model? I compressed it, but the .tar.gz seems to be too big.

Thank you

carrotandjzy, 2022-03-16:
We just need model structure and  dummy input

carrotandjzy, 2022-03-16:
@MarvS95  You can remove pth from tar.gz

MarvS95, 2022-03-16:
Hi @carrotandjzy,

here is my model folder. You can find the dummy input in _dummy_data.py_, where frames of testvideo.mp4 are preprocessed for the model. For quantization I use the _SlowFast_quant.py_ where the function _slowfast(cfg)_ in line 74 builds the model. The model structure is in _SlowFast model structure.txt_ Many thanks for your help.

[slowfast_custom_op.tar.gz](https://github.com/Xilinx/Vitis-AI/files/8260286/slowfast_custom_op.tar.gz)
[SlowFast model structure.txt](https://github.com/Xilinx/Vitis-AI/files/8260377/SlowFast.model.structure.txt)

MarvS95, 2022-03-18:
Hi, I just found out, that the compilation process takes longer until it crashes when I rise the RAM of my virtual machine from 8 GB to 27,5 GB. With 8 GB the process got killed after 3 mins and with 27,5 GB it got killed after 15 mins and I got this error message:

`[UNILOG][FATAL][XCOM_DATA_OUTRANGE][Data value is out of range!] The dynamic space is too small for your data size. Please recheck the bankgroup assigner or tiling process.`

MarvS95, 2022-03-25:
Hi @carrotandjzy,

did you find an error anywhere in the dummy input/ the model?

Thank you

carrotandjzy, 2022-03-28:
Hi @MarvS95 
There is an issue about bias correction. You turn it off by adding cmd option "--nndct_param_corr False".
In your quant.sh :
 python3  SlowFast_quant.py --out-channels 3  --output-dir ./build/quant_model/ 
-->
 python3  SlowFast_quant.py **--nndct_param_corr False** --out-channels 3  --output-dir ./build/quant_model/ 

Thanks

MarvS95, 2022-03-28:
Hi @carrotandjzy 

thank you really much. I added the cmd option to my quantization sh, but still get the same errors :/

When I change the sequence in head_helper.py (line 84) from

```
        x = torch.cat((pool_out0, pool_out1), 1)
        x = x.permute((0, 2, 3, 4, 1))
```
to
```
        pool_out0 = pool_out0.permute((0, 2, 3, 4, 1))
        pool_out1 = pool_out1.permute((0, 2, 3, 4, 1))
        x = torch.cat((pool_out0, pool_out1), 4)
```
the warning
```
[UNILOG][WARNING] xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__22431, type = concat-fix} has been assigned to CPU: [Input xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__AvgPool3d_pathway0_avgpool__22395_inserted_fix_8, type = float2fix} is not in DPU subgraph. And output dimension is not 4.].
[UNILOG][WARNING] xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__22431, type = concat-fix} has been assigned to CPU: [Input xir::Op{name = SlowFastModel__SlowFastModel_ResNetBasicHead_head__AvgPool3d_pathway0_avgpool__22395_inserted_fix_8, type = float2fix} is not in DPU subgraph. And output dimension is not 4.].
```
disappears.

I also found out, that I can quantize and compile half the model. I will try to compile the whole model on a GPU Server using GPU Docker.

closed 2022-03-30 by niuxjxlnx