Xilinx/Vitis-AI #693: Facing cu timeout! issue on kv260 on custom model

910560, 2022-03-04:
Here is the log of the terminal 

root@xilinx-k26-starterkit-2021_2:~/Vitis-AI/demo/VART/deep_wb_pt# xdputil benchmark deepWBnet.xmodel 1
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0304 08:00:47.421594  2168 xrt_cu.cpp:188] cu timeout! device_core_idx 0  handle=0xffffac019190 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10009  bo=1 is_done 0 
I0304 08:00:47.421727  2168 xrt_cu.cpp:99] Total: 10009604us    ToDriver: 18446742216815791us   ToCU: 0us       Complete: 0us   Done: 1866903364us
F0304 08:00:47.421762  2168 dpu_control_xrt_edge.cpp:186] dpu timeout! core_idx = 0
 LSTART 7  LEND 7  CSTART 1  CEND 1  SSTART 0  SEND 0  MSTART 0  MEND 0  CYCLE_L 14358228  CYCLE_H 75 
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  2167 Aborted                 /usr/bin/python3 -m xdputil $*

qianglin-xlnx, 2022-03-05:
Hi @910560 
Before you run deepWBnet.xmodel, could you please run the model from Xilinx model zoo to check if the DPU is working normally, such as resnet50
If yes, could you please share deepWBnet.xmodel file？ Thank you.

910560, 2022-03-06:
yes, both pre-built vitis-ai-library resnet50 xmodel and re-build resnet50 xmodel from docker are working well.

[deepWBnet.zip](https://github.com/Xilinx/Vitis-AI/files/8192325/deepWBnet.zip)

Apart from doing xdputil, I can deserialize the xmodel and get corresponding inputs size and output size. But when i try to execute_async and wait, it also produce the cu timeout.

Thanks

qianglin-xlnx, 2022-03-07:
Hi @910560 
Thank you for sharing the model. 
However, I run the model successfully on my side, as shown below. 
[image]

I use the KV260 2.0 image:  https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-kv260-dpu-v2021.2-v2.0.0.img.gz
So could you run the following to check the environment.
[image]
If you use the VART1.4, then you can upgrade the VART to 2.0.

910560, 2022-03-07:
thanks for your confirmation that the xmodel is working properly

the reason i have the above error is i wrong feed incorrect input shape to the model and cause the dpu somehow hang or what. 
and after that i cannot perform the xdputils benchmark

closed 2022-03-07 by 910560