Xilinx/Vitis-AI #694: no python2 when building GPU docker image

mhanuel26, 2022-03-04:
Hi there,

When I build the GPU docker image I noticed that python2 is not installed. On the default CPU docker image (downloaded it is).

For instance, here is the output of the CPU docker image showing both python versions.

```
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_1.4 > python
Python 3.6.13 | packaged by conda-forge | (default, Sep 23 2021, 07:56:31) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_1.4 > python2
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_1.4 > which python
/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/python
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_1.4 > which python2
/usr/bin/python2
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_1.4 > which python3
/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/python3
```
On my compiled version of the GPU 

```
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_2.0/data > python
Python 3.6.15 | packaged by conda-forge | (default, Dec  3 2021, 18:49:41) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_2.0/data > python2
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_2.0/data > which python2
/usr/bin/python2
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_2.0/data > which python3
/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/python3
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_2.0/data > which python
/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/python
(vitis-ai-caffe) Vitis-AI /workspace/models/AI-Model-Zoo/cf_ssdmobilenetv2_bdd_360_480_6.57G_2.0/data >
```

As you can see python2 is not within the conda environment. 
I am having issues as cf_ssdmobilenetv2_bdd_360_480_6.57G_2.0 data preprocessing script have issues running on python3 (not sure if it worth to open a separate issue for that) .

Any comments?

Thanks,

wangxd-xlnx, 2022-03-09:
Hi @mhanuel26 

Could you help provide some more details? I guess that you mean this script 'data/data_preprocess.sh' and what's the specific error?  It would be best if you could provide screenshots. Thanks.

mhanuel26, 2022-03-09:
Hi @wangxd-xlnx ,

I didn't keep screenshots but if I remember correctly the problem is in file convert_coco2voc_like.py, function generate_annotation_files, for some reason python2 and python3 differ in the way you get the file object, and for instance line 77 where is does 
`category_id = anno_info['category_id']`
was complaining about the object being  a bin not str. 

I end up running that code separately outside the xilinx docker image using python2 wince it does not depend on caffee.

wangxd-xlnx, 2022-03-11:
Hi @mhanuel26 

Please notice that if you could use Vitis-AI docker, you just need to activate the conda env ‘vitis-ai-caffe'. You don't need to compile the caffe-xilinx source code manually. It's just for the environment that is inconvenient to use docker.

The caffe-xilinx is precompiled in conda env ‘vitis-ai-caffe'. You can directly use it. So you could have another try that exit and  run docker but don't compile the caffe-xilinx.

PATH: /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/

[image]

mhanuel26, 2022-03-11:
Hi @wanghy-xlnx ,

The problem is that the script calls a python script called create_annoset.py, first looking into the python folder under 

`/workspace/models/AI-Model-Zoo/caffe-xilinx/python`

when you open a python console like you did inside that directory, the command fail, if you do outside it works. 

I have to manually change the name of that python folder to python_bkp, so the script then looks into the folder 
 /opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/ 

Please take a look at the screenshot below

[image]

wangxd-xlnx, 2022-03-14:
Hi @mhanuel26,

Yes, I understand what you have commented. We suggest that please ignore /workspace/models/AI-Model-Zoo/caffe-xilinx and don't open and use its python scripts. This caffe-xilinx source code is just for users that cannot pull and run docker image.
I noticed that you have running docker and activate env '_vitis-ai-caffe_'. This is enough, so don't need open the caffe-xilinx folder to avoid conflicts.

mhanuel26, 2022-03-14:
Hi @wanghy-xlnx ,

I don't intentionally open that, if you follow the xilinx guide on the readme of this example, the script will try to invoke the code that is on that python folder of Vitis-AI by default. 

Thanks,

hanxue, 2022-03-17:
Hi @mhanuel26 ,

To clarify on the different versions of python, when you are in a Vitis AI conda environment, use the default `python` binary, which you will get from `which python`. It is always in the path `/opt/vitis_ai/conda/envs/vitis-ai-<framework>/bin/python` For example:

```
/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/python
```

Using any other python version will cause incompatibility problems. The sam version of Python is used for building all  Vitis AI tools (for example: caffe quantizer, and compilers). Likewise, all the conda and python packages have been tested with this specific version of python.

qianglin-xlnx, 2022-04-07:
Hi @mhanuel26 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-04-07 by qianglin-xlnx