Xilinx/Vitis-AI #695: Pytorch Quantizer error when compile custom Mobilenetv2 model

SamsonChau, 2022-03-07:
I am trying to deploy my mobilenet v2 model in Vitis AI 1.4, My model is a pytorch model with two input an ir image & depth image to perform classification in different face id. 
The input image will first input to the mobilenetv2 model() then the output will pass put into a LSTM model in my design. 

All models in the pretained and tested in our local machine. However  the model is not being quantize correctly when put in the quantizer, and even after it being quantized (cpu docker verion) and export the x model error happened when compile it for the vck190-production version platform and the ZCU102- DPU(v3.3 same as base trd). 

Here's the version of our tools
model type: pytrorch model - Mobilenet v2(CNN)  - LSTM (RNN)
pytorch version: 1.7.1 cuda 10.2
Vitis AI tools version: 1.4

I have tried to quantize the Mobilenetv2 model with both cpu docker and gpu docker with different results: 
CPU platform
can be quantize but fail when compile
output log:  
```
>>> quantizer = torch_quantizer('calib',model,(depth_img,ir_img),device=device)

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing Mobilenet_v3_dual_input...

[VAIQ_WARN]: ReLU6 has been replaced by ReLU.
/opt/vitis_ai/conda/envs/Pytorch17/lib/python3.6/site-packages/torch/nn/functional.py:1639: UserWarning: nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.
  warnings.warn("nn.functional.sigmoid is deprecated. Use torch.sigmoid instead.")
/workspace/models/Antispoofing/models/mobilenetv3_hari.py:88: TracerWarning: Converting a tensor to a Python index might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  scale = torch.cat([scale]*wh, 2)
/workspace/models/Antispoofing/models/mobilenetv3_hari.py:89: TracerWarning: Converting a tensor to a Python index might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  scale = torch.cat([scale]*wh, 3)

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/Mobilenet_v3_dual_input.py)

(Pytorch17) Vitis-AI /workspace/models/Antispoofing/compiled > vai_c_xir -x ../quantizer_result/Mobilenet_v3_dual_input_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190/arch.json -o ./ -n Mobilnev3_antispoofing
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] Target architecture: DPUCVDX8G_ISA1_C32B3
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCVDX8G_ISA1_C32B3
[UNILOG][INFO] Graph name: Mobilenet_v3_dual_input, with op num: 1049
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XIR_MULTI_DEFINED_OP][Multiple definition of OP!] Mobilenet_v3_dual_input__Mobilenet_v3_dual_input_Sequential_depth_features__InvertedResidual_13__Sequential_block__SqueezeExcitation_2__24763_inserted_fix_245_Mobilenet_v3_dual_input__Mobilenet_v3_dual_input_Sequential_depth_features__InvertedResidual_13__Sequential_block__SqueezeExcitation_2__24778
*** Check failure stack trace: ***
Aborted (core dumped)
(Pytorch17) Vitis-AI /workspace/models/Antispoofing/compiled > vai_c_xir -x ../quantize_result/Mobilenet_v3_dual_input_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190/arch.json -o ./ -n Mobilnev3_antispoofing
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] Target architecture: DPUCVDX8G_ISA1_C32B3
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCVDX8G_ISA1_C32B3
[UNILOG][INFO] Graph name: Mobilenet_v3_dual_input, with op num: 1049
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XIR_MULTI_DEFINED_OP][Multiple definition of OP!] Mobilenet_v3_dual_input__Mobilenet_v3_dual_input_Sequential_ir_features__InvertedResidual_5__Sequential_block__SqueezeExcitation_2__25643_inserted_fix_28_Mobilenet_v3_dual_input__Mobilenet_v3_dual_input_Sequential_ir_features__InvertedResidual_5__Sequential_block__SqueezeExcitation_2__25658

[UNILOG][FATAL][XIR_MULTI_DEFINED_OP][Multiple definition of OP!] Mobilenet_v3_dual_input__Mobilenet_v3_dual_input_Sequential_ir_features__InvertedResidual_4__Sequential_block__SqueezeExcitation_2__25490_inserted_fix_0_Mobilenet_v3_dual_input__Mobilenet_v3_dual_input_Sequential_ir_features__InvertedResidual_4__Sequential_block__SqueezeExcitation_2__25505

```
quantization code:
```
quantizer = torch_quantizer('calib',model,(depth_img,ir_img),device=device)
model_q=quantizer.quant_model
quantizer.export_quant_config()
quantizer = torch_quantizer('test',model,(depth_img,ir_img),device=device)
dump_xmodel("quantizer_result",deploy_check=True)
[... 770 lines cut ...]

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::classifier.3.weight

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::classifier.3.bias

[VAIQ_WARN]: Quantization is not performed completely, check if model inference function is called!!!

```
quantization code:
```
    #### Quantization ####
    print("start quantization calibration")
    quantizer = torch_quantizer('calib', model, input) 
    quantized_model = quantizer.quant_model
    quantizer.export_quant_config()

    input = (torch.randn(1, 3, 224, 224))
    quantizer_test = torch_quantizer('test', model, input) 
    quantized_test_model = quantizer_test.quant_model
    quantizer_test.export_xmodel(deploy_check=False)
```

These are the log, is my model not support by the quantizer or I put the wrong dpu design in the compile cause the error? 

Also is that the RNN model are not support in the current vck190 DPU ip?  Is the only way is run in to cpu?

niuxjxlnx, 2022-03-08:
Hi, @SamsonChau:

Please don't call torch_quantizer() multiple times in model script. The code should right as the following:
    quantizer = torch_quantizer('calib', model, input) 
    quantized_model = quantizer.quant_model
    **quantized_model evaluation code**
    quantizer.export_quant_config()
Then the model will be quantized in calibraion process. 

You can refer to the example 
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py 
and read 
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/README.md 
for how to run it.

SamsonChau, 2022-03-10:
Thanks for reply, 

I have change the quantize part in to the following, 
```python
import os
import sys
import numpy as np
from PIL import Image
import random

import torch
import torchvision
import torch.nn as nn
import torch.optim
import torch.nn.functional as F
from torchvision import transforms
from torch.autograd.variable import Variable

from models import mobilenetv3_hari //my model 
from pytorch_nndct import QatProcessor
from pytorch_nndct.apis import torch_quantizer, dump_xmodel

def quantize():
    # use GPU if available   
    device = torch.device('cuda' if torch.cuda.is_available() else "cpu")
   
    # load trained model
    model = mobilenetv3_hari.mobilenet_v3_large_dual_input(pretrained=True).to(device)
    model.eval()
    
    # load the data set 
    depth_list = []
    ir_list = []
    img_size =224

    normalize = transforms.Normalize(mean=[0.485],
                                     std=[0.229])
    img_size = 224
    transform = transforms.Compose([
            transforms.Resize((img_size, img_size)),
            transforms.ToTensor(),
            normalize,
        ])
    f = open('live_file_list.txt', 'r').read().splitlines()
    for file in f:
        if 'depth' in file:
            depth_list.append(file)
        elif 'ir' in file:
            ir_list.append(file)

    for i in range(0, len(depth_list)):
        print(i)
        with torch.no_grad():
            depth_img = Image.open(depth_list[i]).convert('L')
            depth_img = transform(depth_img)
            depth_img = depth_img.unsqueeze(0)
            depth_img = Variable(depth_img).float().to(device)
            
            ir_image = Image.open(ir_list[i]).convert('L')
            ir_image = transform(ir_image)
            ir_img = ir_image.unsqueeze(0)
            ir_img = Variable(ir_img).float().to(device)
            output = model(depth_img, ir_img)
    #set the data set in input 
    input = (depth_img,ir_img)    
   
    #### Quantization ####
    print("start quantization calibration")
    quantizer = torch_quantizer('calib', model, input) 
    quantized_model = quantizer.quant_model
[... 961 lines cut ...]

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::ir_features.15.block.2.fc2.weight

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::ir_features.15.block.2.fc2.bias

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::ir_features.15.block.3.0.weight

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::ir_features.15.block.3.0.bias

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::ir_features.16.0.weight

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::ir_features.16.0.bias

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::classifier.0.weight

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::classifier.0.bias

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::classifier.3.weight

[VAIQ_WARN]: Parameter tensor is not quantized: Mobilenet_v3_dual_input::classifier.3.bias

[VAIQ_WARN]: Quantization is not performed completely, check if model inference function is called!!!
```

So is the xilinx quantizer not support the operater inside my model or there have sone issue regarding to the warning ```[VAIQ_WARN]: ReLU6 has been replaced by ReLU.```  ?

SamsonChau, 2022-03-10:
Hi @niuxjxlnx 

Thanks for your solution, is this what you mean about the model evaluation function? 
 
```python
    print("start quantization calibration")
    quantizer = torch_quantizer('calib', model, input) 
    quantized_model = quantizer.quant_model
   #model evalution function
    quantized_model = quantized_model.to(device)
    quantized_model.eval()
    quantizer.export_quant_config()
   
  # some code to test the accuracy here of quantized code
```
I have tested the code above and there still have all warning, Is there any wrong code?

niuxjxlnx, 2022-03-10:
Hi, @SamsonChau :
quantized_model.eval() is just to set a flag, not to do model evalution. 

For example, in https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py,  the following code is to do model evaluation:

```
  # record  modules float model accuracy
  # add modules float model accuracy here
  acc_org1 = 0.0
  acc_org5 = 0.0
  loss_org = 0.0

  #register_modification_hooks(model_gen, train=False)
  acc1_gen, acc5_gen, loss_gen = evaluate(quant_model, val_loader, loss_fn)

  # logging accuracy
  print('loss: %g' % (loss_gen))
  print('top-1 / top-5 accuracy: %g / %g' % (acc1_gen, acc5_gen))

```

closed 2022-03-14 by niuxjxlnx