Xilinx/Vitis-AI #696: The prototext file is missing when deploying the Reid model

jzymessi, 2022-03-07:
I am now trying to train Reid model myself. At present, I have quantified, compiled and deployed the trained model. However, when deploying the model, it is found that there is no peotxt file. So I put Pt in model zoo_ personreid-res50_ market1501_ 256_ 128_ 5.4G_ The prototext file of the 2.0 model is directly copied .
[image]
However, compared with the accuracy results of the test, the accuracy loss is large. According to my understanding, the prototext file is the pre-processing file of the model. The model I trained is inconsistent with the official pre-processing method, which may lead to the loss of accuracy. Can the prototext file be generated in other ways? If not, can you tell me if the parameters in this file are obtained?
Looking forward to your reply, thanks.

yongshengxu-xlnx, 2022-03-08:
Hi, @jzymessi 
This file data is used for pre- and post-processing, where mean and scale are pre-processing data, and the image will be subtracted from the mean and multiplied by the scale. The parameters with bn in the back are used for post-processing bn operations, and are parameters obtained for a specific model. They are used in: [https://github.com/Xilinx/Vitis-AI/blob/ba8905493876674716ab08ea4266962193df970b/tools/Vitis-AI-Library/xnnpp/src/reid.cpp#L73](https://github.com/Xilinx/Vitis-AI/blob/ba8905493876674716ab08ea4266962193df970b/tools/Vitis-AI-Library/xnnpp/src/reid.cpp#L73)
This parameter needs to be obtained in your own model.

jzymessi, 2022-03-08:
> 嗨，@jzymessi此文件数据用于预处理和后处理，其中平均值和比例是预处理数据，图像将从平均值中减去并乘以比例。背面带有 bn 的参数用于后处理 bn 操作，并且是为特定模型获得的参数。它们用于：
> 
> https://github.com/Xilinx/Vitis-AI/blob/ba8905493876674716ab08ea4266962193df970b/tools/Vitis-AI-Library/xnnpp/src/reid.cpp#L73
> 
> 
> 此参数需要在您自己的模型中获取。

@yongshengxu-xlnx 

> Hi, @jzymessi This file data is used for pre- and post-processing, where mean and scale are pre-processing data, and the image will be subtracted from the mean and multiplied by the scale. The [...]

@yongshengxu-xlnx  Thank you for your reply.
But I have some questions:
1.Can you tell me a flow about how to generate BN these parameters?
2.My pre-processing of pictures is inconsistent with the C pre-processing of Vitis AI. Do I need to adjust the pre-processing process during deployment? If necessary, what changes need to be made?
Looking forward to your reply, thanks.

yongshengxu-xlnx, 2022-03-08:
@jzymessi 
The bn parameter needs to be found in your floating model. Specifically, you need to find out where you did the bn operation, obtain the bn parameter in your original code, and then save it.
You can refer to the sample code in yolov3 for how to modify the preprocessing, like:
[https://github.com/Xilinx/Vitis-AI/blob/ba8905493876674716ab08ea4266962193df970b/tools/Vitis-AI-Library/yolov3/src/yolov3_imp.cpp#L53](https://github.com/Xilinx/Vitis-AI/blob/ba8905493876674716ab08ea4266962193df970b/tools/Vitis-AI-Library/yolov3/src/yolov3_imp.cpp#L53)

LorenzoSun-V, 2022-03-09:
hi, @yongshengxu-xlnx .
I've extracted BN params from my own model. But the BN params seem to be consisted of 'bn_means', 'bn_vars' and 'bn_weights' in ReID prototxt Xilinx provided while the BN params in my own model are like: {"backbone.bn1.running_mean":tensor[... , ... , ...]}（tensor shape is decided by the channel numbers）,{'backbone.layer0.1.bn2.mean':tensor[... , ... , ...]}. How should I extract my BN params so that the cpp code can load my BN params correctly? Because your BN params are quite different from mine. Would you please provide the code extracting BN params from pth to prototxt?

LorenzoSun-V, 2022-03-09:
[image]
[image]
[image]
These are the part of my BN params.

yongshengxu-xlnx, 2022-03-09:
@LorenzoSun-V 
I just print BN params line by line to file，then add prefixes in batches. 
Then copy the val to prototxt.

jzymessi, 2022-03-15:
> @LorenzoSun-V I just print BN params line by line to file，then add prefixes in batches. Then copy the val to prototxt.

We have tried the above methods, but the final accuracy measured by using our own BN parameters is not as good as the official BN parameters. Do you have the relevant code to extract BN parameters for our reference? Thank you.

LorenzoSun-V, 2022-03-18:
> 

I extract BN params using following code:
`def load_bn_param(weight_path):
    ckpt_state_dict = torch.load(weight_path).state_dict()
    mean_dict = {}
    var_dict = {}
    weight_dict = {}
    for k in list(ckpt_state_dict.keys()):
        if 'bn' in k:
            if 'mean' in k:
                mean_dict[k] = ckpt_state_dict[k].cpu().numpy()
            elif 'var' in k:
                var_dict[k] = ckpt_state_dict[k].cpu().numpy()
            elif 'weight' in k:
                weight_dict[k] = ckpt_state_dict[k].cpu().numpy()
    return mean_dict, var_dict, weight_dict

def save_bn_param(mean_dict, var_dict, weight_dict, reid_param_path):
    with open(reid_param_path, 'w+') as f:
        for k in mean_dict:
            for i in mean_dict[k]:
                f.write(f'bn_means:{i}\n')
        for k in var_dict:
            for i in var_dict[k]:
                f.write(f'bn_vars:{i}\n')
        for k in weight_dict:
            for i in weight_dict[k]:
                f.write(f'bn_weights:{i}\n')
    print(f'ReID params have been save to {reid_param_path}')

def main(weight_path, reid_param_path):
    mean_dict, var_dict, weight_dict = load_bn_param(weight_path)
    save_bn_param(mean_dict, var_dict, weight_dict, reid_param_path)

if __name__ == "__main__":
    weight_path = "/workspace/lorenzo/ReID/xilinx_reid_quantity/float/Market1501_BoT_R50.pth"
    reid_param_path = '/workspace/lorenzo/ReID/xilinx_reid_quantity/float/Market1501_BoT_R50.txt'
    main(weight_path, reid_param_path)`

However, my BN params are more than yours. Would you please provide the code to extract BN params in the pth?

LorenzoSun-V, 2022-03-18:
I extract BN params using following code:
def load_bn_param(weight_path):
ckpt_state_dict = torch.load(weight_path).state_dict()
mean_dict = {}
var_dict = {}
weight_dict = {}
for k in list(ckpt_state_dict.keys()):
if 'bn' in k:
if 'mean' in k:
mean_dict[k] = ckpt_state_dict[k].cpu().numpy()
elif 'var' in k:
var_dict[k] = ckpt_state_dict[k].cpu().numpy()
elif 'weight' in k:
weight_dict[k] = ckpt_state_dict[k].cpu().numpy()
return mean_dict, var_dict, weight_dict

def save_bn_param(mean_dict, var_dict, weight_dict, reid_param_path):
with open(reid_param_path, 'w+') as f:
for k in mean_dict:
for i in mean_dict[k]:
f.write(f'bn_means:{i}\n')
for k in var_dict:
for i in var_dict[k]:
f.write(f'bn_vars:{i}\n')
for k in weight_dict:
for i in weight_dict[k]:
f.write(f'bn_weights:{i}\n')
print(f'ReID params have been save to {reid_param_path}')

def main(weight_path, reid_param_path):
mean_dict, var_dict, weight_dict = load_bn_param(weight_path)
save_bn_param(mean_dict, var_dict, weight_dict, reid_param_path)

if name == "main":
weight_path = "/workspace/lorenzo/ReID/xilinx_reid_quantity/float/Market1501_BoT_R50.pth"
reid_param_path = '/workspace/lorenzo/ReID/xilinx_reid_quantity/float/Market1501_BoT_R50.txt'
main(weight_path, reid_param_path)

However, my BN params are more than yours. Would you please provide the code to extract BN params in the pth?

LorenzoSun-V, 2022-03-18:
> @LorenzoSun-V I just print BN params line by line to file，then add prefixes in batches. Then copy the val to prototxt.

I extract BN params using following code:
[image]
However, my BN params are more than yours. Would you please provide the code to extract BN params in the pth?

yongshengxu-xlnx, 2022-03-18:
@LorenzoSun-V I will ask for algorithm colleague to help, the question will be answered later.

wenjingk-xilinx, 2022-03-18:
Hi @LorenzoSun-V,

Firstly, I need to explain why there we need to do bn in the post-processing. There are a lot of bn layers used in the Resnet network, these bn layers in the network are put after conv layer, except the last bn layer. The bn layers after conv layer are merged into conv layer, except the last bn layer.  So we put the last bn layer in the post-processing. 

Secondly, please have a check on your own network. Do you have a sole bn layer at the end of the network? If you don't have such a bn layer, you don't need to do this post-processing. If do have, then just get its parameters from the checkpoint file. From the previous conversation, It seems that you are not sure about the layer's name, I suggest you print your network to check the layer's names. Then you will know which bn is the one you need. Obviously, names like 'backbone.layer0.1.bn2.mean' are not the last bn layer. Another way is to check the dimension of bn params, it should be consistent with the dimension of your final output feature. 

Hope these are helpful to you. Have a try again.

wenjingk-xilinx, 2022-03-18:
If you are still not sure, please show me the code of how you built your model.

LorenzoSun-V, 2022-03-18:
> [image] [image] [image] These are the part of my BN params.

The model in picture was trained on the project 'fast-reid' created by JD Team.  Since the pth can't be pruned and quantitied by the tools provided by Vitis-AI, I trained my reid task based on 'reid-strong-baseline' created by Hao Luo in Zhejiang University

LorenzoSun-V, 2022-03-18:
[https://github.com/michuanhaohao/reid-strong-baseline/blob/master/modeling/baseline.py](https://github.com/michuanhaohao/reid-strong-baseline/blob/master/modeling/baseline.py)
I use this code to build my model.

LorenzoSun-V, 2022-03-18:
> Hi @LorenzoSun-V,
> 
> Firstly, I need to explain why there we need to do bn in the post-processing. There are a lot of bn layers used in the Resnet network, these bn layers in the network are put after conv layer, except the last bn layer. The bn layers after conv layer are merged into conv layer, except the last bn layer. So we put the last bn layer in the post-processing.
> 
> Secondly, please have a check on your own network. Do you have a sole bn layer at the end of the network? If you don't have such a bn layer, you don't need to do this post-processing. If do have, then just get its parameters from the checkpoint file. From the previous conversation, It seems that you are not sure about the layer's name, I suggest you print your network to check the layer's name. Then you will know which bn is the one you need. Obviously, names like 'backbone.layer0.1.bn2.mean' are bn layers in the backbone, not the last sole one. Another way is to check the dimension of bn params, it should be consistent with the dimension of your final output feature.
> 
> Hope these are helpful to you. Have a try again.

I seem to understand what u said, I'll try again. Thx a lot.

LorenzoSun-V, 2022-03-18:
> Hi @LorenzoSun-V,
>
> Firstly, I need to explain why there we need to do bn in the post-processing. There are a lot of bn layers used in the Resnet network, these bn layers in the network are put after [...]

I fixed my code and restore the accuracy successfully. Thanks!!  :-D

wenjingk-xilinx, 2022-03-18:
Great! Please help to close the issue. :-D

closed 2022-03-18 by jzymessi