Xilinx/Vitis-AI #697: resnet_dump.prototxt missing

PrashanthHC16, 2022-03-07:
I am running the example on /examples/DPUCZDX8G/ofa_resnet50.

The procedure instructs to create a resnet50_fp2000.prototxt from resnet50_dump.txt

After compiling, i dont find the prototxt file, and get the error : 

F0312 23:02:42.137238  1074 configurable_dpu_task_imp.cpp:117] cannot find ./resnet50_fp2000/resnet50_fp2000.prototxt
*** Check failure stack trace: ***
Aborted

after running  _./test_jpeg_classification resnet50_fp2000 sample_classification.jpg_

Please help me find the prototxt file, am i missing something in the example process?

yongshengxu-xlnx, 2022-03-08:
Hi, @PrashanthHC16 
You need to copy the  resnet50_fp2000.prototxt you created to ./resnet50_fp2000/resnet50_fp2000.prototxt
modify:
cp resnet50_dump.prototxt resnet50_fp2000.prototxt
vim resnet50_fp2000.prototxt
```
  at line 2: modify the following
          from:    name : "resnet50_dump"
          to:    name : "resnet50_fp2000"
```
copy:
```
  [Host]cd /workspace/ofa/compiled
  [Host]scp resnet50_fp2000.xmodel userid@IP_OF_BOARD:/home/root/Vitis-AI/demo/Vitis-AI-Library/samples/classification/resnet50_fp2000/
  [Host]scp resnet50_fp2000.prototxt userid@IP_OF_BOARD:/home/root/Vitis-AI/demo/Vitis-AI-Library/samples/classification/resnet50_fp2000/
```
Make sure the name is correct and the file is in the path.

PrashanthHC16, 2022-03-08:
Thanks for the reply. 
Only the following three files are generated after compilation : 

md5sum.txt
meta.json
resnet50_fp2000.xmodel

The prototxt file referred in example is not generated. The manual on pyTorch compilation also says these files will be generated. However caffe document says prototxt will also be generated.

But the example uses pyTorch.

yongshengxu-xlnx, 2022-03-08:
@PrashanthHC16 
You can download the xmodel and prototxt at this link: [https://www.xilinx.com/bin/public/openDownload?filename=resnet50-zcu102_zcu104_kv260-r2.0.0.tar.gz](https://www.xilinx.com/bin/public/openDownload?filename=resnet50-zcu102_zcu104_kv260-r2.0.0.tar.gz)
Then you copy the resnet50.pototxt to resnet50_fp2000.prototxt and modify.

PrashanthHC16, 2022-03-08:
It worked. Thank you @yongshengxu-xlnx.

One more query : OFA also supports mobilenetv3. How can I use mobilenetV3 models generated from OFA, using the same example flow? What are the Vitis-AI specific modifications needed in ofa_resnet50_search_byOps.ipnyb, ofa_quant.py.

Thanks in advance, I'm closing the issue.

closed 2022-03-08 by PrashanthHC16

yongshengxu-xlnx, 2022-03-08:
We are not good at ofa, please @taeheej  to answer.