Xilinx/Vitis-AI #701: Is why?

Anas-liu, 2022-03-09:
I use VPN, tried many times, just error. ubuntu18.4, build GPU docker

Google Mock not present.  Fetching gmock-1.7.0 from the web...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:02:10 --:--:--     0curl: (7) Failed to connect to github.com port 443: Connection timed out
The command '/bin/bash -c cd /tmp; wget --progress=dot:mega https://codeload.github.com/google/protobuf/zip/v3.4.0     && unzip v3.4.0     && cd protobuf-3.4.0     && ./autogen.sh     && ./configure     && make -j     && make install     && ldconfig     && rm -fr /tmp/*' returned a non-zero code: 7

hanxue, 2022-03-17:
Hi @Anas-liu ,

Can you check your if you can download the protobuf zip file from Github? 

From the error message seems that your host has issue connecting to Github server. You can check by directly running the wget command

```
$ wget https://codeload.github.com/google/protobuf/zip/v3.4.0
--2022-03-16 21:14:56--  https://codeload.github.com/google/protobuf/zip/v3.4.0
Resolving codeload.github.com (codeload.github.com)... 140.82.112.10
Connecting to codeload.github.com (codeload.github.com)|140.82.112.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘v3.4.0’

    [      <=>                              ] 6,330,829   4.75MB/s   in 1.3s   

2022-03-16 21:14:58 (4.75 MB/s) - ‘v3.4.0’ saved [6330829]
```

qianglin-xlnx, 2022-06-11:
Hi @Anas-liu 
Has the problem been solved?

qianglin-xlnx, 2022-06-23:
Hi @Anas-liu 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-06-23 by qianglin-xlnx