Xilinx/Vitis-AI #702: yolov4-tiny  with 7 categories VS  yolov4-tiny with 3 categories

charles-str, 2022-03-09:
Due to yolov4-tiny CPSdarknet has strided_slice, vitis ai does not support,
I replace strided_slice with 1x1  convolution， but when I train yolov4-tiny with 3 categories, the speed has 30 fps,
yolov4-tiny  with 7 categories has only 3 fps.
this is the netron of with 7 categories.
[image]
this is the netron of with 3 categories.
[image]

charles-str, 2022-03-09:
I train yolov4-tiny with darknet

closed 2022-03-09 by charles-str

tuan3ihust, 2022-11-27:
Hello did you solve the problem?
I trained yolov4-tiny with darknet too, and convert to caffe, but the compiled model have multiple DPU subgraph (4), I think it is because vitis ai doesn't support route layer with groups, how do you solve it? did changing it to 1x1 conv reduce model accuracy? thank you.
[image]