Xilinx/Vitis-AI #703: how do I get layer name to write to prototxt file？

worldback, 2022-03-09:
yolov3 layer name in yolov3.prototxt is 59,67,75。 How are these three numbers determined?  how to determined yolov5 layer name?

yongshengxu-xlnx, 2022-03-09:
Hi, @worldback 
You can use xir subgraph $your_xmodel_name to see the outputs layer name. like:
`xir subgraph yolov4_leaky_spp_m.xmodel |grep DPU`
you will see like :
`subgraph_layer0-conv [device=DPU,fingerprint=0x1000020f6014407,DPU=DPUCZDX8G_ISA0_B4096_MAX_BG2,I=[xir_tensor{data_fixed:(1,416,416,3), fixpos=7 # of elements= 519168}],O=[xir_tensor{layer138-conv_fixed:(1,52,52,255), fixpos=4 # of elements= 689520},xir_tensor{layer149-conv_fixed:(1,26,26,255), fixpos=4 # of elements= 172380},xir_tensor{layer160-conv_fixed:(1,13,13,255), fixpos=4 # of elements= 43095}]]`
the output layer name is : layer138-conv_fixed, layer149-conv_fixed, layer160-conv_fixed. So in the portotxt  the layer name will set to 138, 149, 169.

worldback, 2022-03-09:
subgraph_Model__Model_Bottleneck_model__Bottleneck_11__Conv_cv1__Conv2d_conv__input_106 [device=DPU,fingerprint=0x1000020f6014404,DPU=DPUCVDX8G_ISA0_B1600_01000020F6014404,I=[xir_tensor{Model__input_0_fix:(1,416,416,3), fixpos=6 # of elements= 519168}],O=[xir_tensor{Model__Model_Detect_model__Detect_28__Conv2d_m__ModuleList_2__9686_fix:(1,13,13,36), fixpos=1 # of elements= 6084},xir_tensor{Model__Model_Detect_model__Detect_28__Conv2d_m__ModuleList_0__9592_fix:(1,52,52,36), fixpos=2 # of elements= 97344},xir_tensor{Model__Model_Detect_model__Detect_28__Conv2d_m__ModuleList_1__9639_fix:(1,26,26,36), fixpos=1 # of elements= 24336}]]
It is the output xir subgraph yolov3.xmodel |grep DPU。Is that normal ？

yongshengxu-xlnx, 2022-03-10:
@worldback 
Because I don't know your network structure, I can't be sure whether your output is correct. These three outputs:
`[xir_tensor{Model__Model_Detect_model__Detect_28__Conv2d_m__ModuleList_2__9686_fix:(1,13,13,36), fixpos=1 # of elements= 6084},xir_tensor{Model__Model_Detect_model__Detect_28__Conv2d_m__ModuleList_0__9592_fix:(1,52,52,36), fixpos=2 # of elements= 97344},xir_tensor{Model__Model_Detect_model__Detect_28__Conv2d_m__ModuleList_1__9639_fix:(1,26,26,36), fixpos=1 # of elements= 24336}]]`
 are the output layers found by xir. You can compare the name and shape of the network output you are using. If they match, you can try to use these three outputs for subsequent processing to see if can get expected result.
The layer name in prototxt is set by your output name, You can set it to the full name of the output name, or you can set the layer-specific serial number, like 9592, 9639,  9686

closed 2022-03-17 by worldback