Xilinx/Vitis-AI #704: yolov3 effect  different after compiled

worldback, 2022-03-09:
I trained a Yolov3 model from https://github.com/ultralytics/yolov3/。This is the effect before compiled，
[image]
after compiled model， the effect is
[image]。
[image]
Does anyone know what is the reason or how to troubleshoot?

zz002, 2022-03-10:
@worldback Which model library are you using to run the model 。

worldback, 2022-03-10:
@zz002  1.3

[image]

and The model compilation environment is

[image]

zz002, 2022-03-10:
@worldback 我 I didn't find the yolo.cpp you reported wrong in the 1.3.2 codebase. I can only tell you about the way to solve this kind of problem.
1.The quantitative model will output a golden, run the dpu subgraph part of the model separately, pass in golden, compare the output,
2.（1.） Check that there is no problem, step by step check the pre- and post-processing of the model to see what is not as expected

qianglin-xlnx, 2022-04-14:
Hi @worldback 
Is this issue solved?

qianglin-xlnx, 2022-04-24:
Hi @worldback 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-04-24 by qianglin-xlnx