Xilinx/Vitis-AI #709: TF2 QAT RTX 3090

samagalhaes, 2022-03-14:
I am trying to perform QAT for a functional model in TensorFlow 2, but TF is complaining about PTXAS. What is this? What is missing?

```
[VAI INFO] Start Quantize Calibration...
2022-03-14 17:01:05.093157: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:185] None of the MLIR Optimization Passes are enabled (registered 2)
2022-03-14 17:01:19.984933: I tensorflow/stream_executor/cuda/cuda_dnn.cc:369] Loaded cuDNN version 8201
2022-03-14 17:01:20.939853: I tensorflow/core/platform/default/subprocess.cc:304] Start cannot spawn child process: No such file or directory
2022-03-14 17:01:20.941678: I tensorflow/core/platform/default/subprocess.cc:304] Start cannot spawn child process: No such file or directory
2022-03-14 17:01:20.941699: W tensorflow/stream_executor/gpu/asm_compiler.cc:77] Couldn't get ptxas version string: Internal: Couldn't invoke ptxas --version
2022-03-14 17:01:20.942205: I tensorflow/core/platform/default/subprocess.cc:304] Start cannot spawn child process: No such file or directory
2022-03-14 17:01:20.942274: W tensorflow/stream_executor/gpu/redzone_allocator.cc:314] Internal: Failed to launch ptxas
Relying on driver to perform ptx compilation. 
Modify $PATH to customize ptxas location.
This message will be only logged once.

```

Applepi, 2023-04-23:
Still getting this issue on VAI 2.5 GPU docker image

renjiangyong, 2023-09-14:
Closing since no activity for more than 2 weeks, please reopen if you still have question, thanks

closed 2023-09-14 by renjiangyong