Xilinx/Vitis-AI #711: nndct_kernels.so: cannot open shared object file: No such file or directory

worldback, 2022-03-15:
我在conda虚拟环境中安装torch（我在conda虚拟环境中安装torch（）后，在vai_q_pytorch/pytorch_binding目录执行python setup.py install 命令，）后，在vai_q_pytorch/pytorch_binding目录执行python setup.py install 命令，然后python -c "import pytorch_nndct",报如下错误
[image]

niuxjxlnx, 2022-03-16:
Hi, @worldback:

Per the warning message, your gcc compiler is not compatible with the torch version you installed, so as to vai_q_pytorch is not compiled successfully. 
What is your OS version and gcc version?

worldback, 2022-03-17:
@niuxjxlnx  gcc : 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC)
centos:   CentOS Linux release 7.9.2009 (Core)

This is the output of cat /etc/version:
Linux version 3.10.0-1160.24.1.el7.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC) ) #1 SMP Thu Apr 8 19:51:47 UTC 2021

niuxjxlnx, 2022-03-17:
@worldback:

From the warning message reported by torch/utils/cpp_extension.py, your gcc is too old for pytorch. Some choices you can try:
1) install gcc in you conda environment. Gcc version >=5.4 works for pytorch 1.9 as I know. The re-compiler vai_q_pytorch.
2) Recomipile pytorch from source code and re-compile vai_q_pytorch
3) Start a Ubuntu(version > 16) docker and works in it.

closed 2022-03-30 by niuxjxlnx