Xilinx/Vitis-AI #713: vai_q_pytorch installed from source code fails to load if PetaLinux SDK environment is sourced

lino-alves, 2022-03-16:
Installed vai_q_pytorch from source code as described in 
[https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#install-from-source-code](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#install-from-source-code)

When I run
`python -c "import pytorch_nndct"`
after the PetaLinux SDK environment is sourced
`source $HOME/petalinux_sdk_2021.2/environment-setup-cortexa72-cortexa53-xilinx-linux`
I get the error
`subprocess.CalledProcessError: Command '['which', 'aarch64-xilinx-linux-g++  -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/myusername/petalinux_sdk_2021.2/sysroots/cortexa72-cortexa53-xilinx-linux']' returned non-zero exit status 1.`
but if the SDK is not sourced everything works fine and I get
`[VAIQ_NOTE]: Loading NNDCT kernels...`

Is this expected behaviour?

niuxjxlnx, 2022-03-17:
The failure is because of runtime compilation of vai_q_pytorch. It has some requirement on OS and some other packages:
1. gcc and its version is compatible with version of pytorch used.
2. some compilation tools like ninja/torch cpp_extension.

If the requirement cannot be satisfied, there is another alternative:
When install vai_q_pytorch from source code, not use command line
`python setup.py install`
But use command line:
`python setup.py bdist`
to create a completely built distribution and install the wheel created, which can avoid runtime compilation.

lino-alves, 2022-03-17:
I created a new conda environment from scratch to test and the issue is the same with
`python setup.py bdist`

If the SDK environment is sourced I get the error when I import the pytorch_nndct module, if it is not then everything is fine.
I get the same error to run the above command line if the SDK is sourced, so to install the SDK also cannot be sourced.

niuxjxlnx, 2022-03-17:
@lino-alves :

Did you install the wheel created by "python setup.py bdist" into your conda environment?

lino-alves, 2022-03-17:
I just ran "python setup.py bdist", which now I realize is not enough to install it, so I would need a bit more help here.
I can't find the wheel file, where is it located? Or what commands do I need to run next?

niuxjxlnx, 2022-03-17:
@lino-alves :

Please run "python setup.py bdist_wheel -d ./". Then a wheel file will be created in current directory. You can install it.

lino-alves, 2022-03-17:
I created the wheel file and installed it with pip in a fresh conda environment, but the issue persists.

The error appears to be related to the execution of the "which" command or am I missing something?
If I run the command manually:
`which aarch64-xilinx-linux-g++ -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/myusername/petalinux_sdk_2021.2/sysroots/cortexa72-cortexa53-xilinx-linux`
I get:
`/home/myusername/petalinux_sdk_2021.2/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++`

To me it appears to be fine, but for some reason it does not run well when importing the "pytorch_nndct" module.

qianglin-xlnx, 2022-03-18:
HI @lino-alves 
There is no need to enable "Petalinux SDK" when you do the pytorch model quantization.
If you want to cross-compile the application program, you can open another terminal to enable "Petalinux SDK".

lino-alves, 2022-03-18:
Hi @qianglin-xlnx 

> There is no need to enable "Petalinux SDK" when you do the pytorch model quantization.

That is I have been doing and it is fine, but for me it felt a bit strange that there is a conflict between the tools.

You can close the issue if you feel this is the expected behaviour.

Thanks for your time.

qianglin-xlnx, 2022-03-18:
@lino-alves 
Anyway, thank you for your feedback.

closed 2022-03-18 by qianglin-xlnx