Xilinx/Vitis-AI #714: vaitrace nm: command not found

ash8327, 2022-03-16:
Hi, 

I modified the DPU following the Zynq Ultrascale+ MPSoC DPU TRD Vitis 2021.2 guide. 
I added the element multiply setting and built the project successfully. I also installed the vitis ai libraries as found in UG 1354. 

After compiling my model for the new arch with arch.json I can run my model. However, when I attempt to use vaitrace -p with my python script I get this error: 
sh: nm: command not found
ERROR:root:[nm --version] not exists, please check the document

Additionally when I attempt to benchmark my model with xdputil I get this error:
F0316 20:43:37.201404  1939 dpu_runner.cpp:266] [UNILOG][FATAL][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] cannot open library! lib=libvart-cpu-runner.so, error=ERROR CODE: libvart-cpu-runner.so: cannot open shared object file: No such file or directory
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  1938 Aborted                 /usr/bin/python3 -m xdputil $*

I have recompiled one of the resnet examples and also have attempted to benchmark it with xdputil but that results in the system freezing requiring a reboot.

li900309, 2022-03-17:
hi @ash8327 , I think, there are some modules are required to be enabled in petalinux , e.g the **nm** is in packagegroup-petalinux-self-hosted
https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Profiler#get-started-with-vitis-ai-profiler

jx991220, 2022-03-29:
I had the same problem as you, and now I have successfully solved it. My solution is as follows:
Solution1. Execute "dnf Install packagegroup-petalinux-self-hosted" on the board, but this will download many files and take some time, so it is recommended for Solution2.
Solution2. Execute "dnf install binutils" and "dnf install ldd" sequentially on the board.

ash8327, 2022-03-29:
I was able to include the packagegroup-petalinux-self-hosted and run vaitrace. However, when I attempt to run xdputil benchmark on my custom model I still get the error:

F0329 17:15:53.583701  6666 dpu_runner.cpp:266] [UNILOG][FATAL][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] cannot open library! lib=libvart-cpu-runner.so, error=ERROR CODE: libvart-cpu-runner.so: cannot open shared object file: No such file or directory
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  6662 Aborted                 /usr/bin/python3 -m xdputil $*

I can run xdputil benchmark on the Xilinx provided resnet50. My model isn't much different from resnet.

zz002, 2022-06-16:
@ash8327 Has the problem been solved, if not, can you send me the instructions for testing the model? If there is no reply, the issue will be closed after one week.

closed 2022-06-23 by qianglin-xlnx

gunertatar, 2022-11-27:
Hello everyone 
I m using Kria KV260 vision board and i executed the tf_resne50 example from the Vitis AI VART. I d like to look at the performance of behavior of the board and used the vaitrace ./resnet50 resnet50.xmodel. After execution i got only " profile_summry.csv, summary.csv, vart_trace.csv, vitis_ai_profile.summary, xrt.run_summary" files. However, when i look around i think, xclbin.ex.run_summary file also should be in the executed file. By the way Vitis unifed software platform and vitis_analyzer version 2021.1.
Is there anyone could help me please? I really need to check all files belongs to execution such as HAL_API Calls, OpenCL data, Data Transfer, DPU, buffer exchange and so on.

Thanks