Xilinx/Vitis-AI #716: I have some problem: Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103

siniGAL, 2022-03-18:
I have some problem:
```bash
make[2]: Leaving directory '/tmp/protobuf-3.4.0/src'
Makefile:1488: recipe for target 'all-recursive' failed  
make[1]: Leaving directory '/tmp/protobuf-3.4.0'
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
Makefile:1395: recipe for target 'all' failed
The command '/bin/bash -c cd /tmp; wget --progress=dot:mega https://codeload.github.com/google/protobuf/zip/v3.4.0     && unzip v3.4.0     && cd protobuf-3.4.0     && ./autogen.sh     && ./configure     && make -j     && make install     && ldconfig     && rm -fr /tmp/*' returned a non-zero code: 2  
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103  
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
```
[report.txt](https://github.com/Xilinx/Vitis-AI/files/8282948/report.txt)

_Originally posted by @siniGAL in https://github.com/Xilinx/Vitis-AI/issues/607#issuecomment-1070518836_

Werterror, 2022-08-09:
I highly doubt it was RAM issu
[image]
e

YuetongFang, 2022-08-22:
replace the "`make -j`" with "`make -j$(nproc)`", can solve the problem.

janifer112x, 2022-09-05:
Hi @YuetongFang,
Thanks for your information, we will update the DockerFile accordingly.

closed 2022-09-05 by janifer112x