Xilinx/Vitis-AI #717: 2.0 fails to build with gcc 11.2.0

alfonsosanchezbeato, 2022-03-18:
While building with gcc 11.2.0 I see the following errors:

```
[  6%] Building CXX object util/CMakeFiles/util.dir/src/zero_copy_helper.cpp.o
In file included from /usr/include/xir/attrs/attrs.hpp:23,
                 from /usr/include/xir/op/op.hpp:19,
                 from /home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/util/include/vart/xir_helper.hpp:21,
                 from /home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/util/src/xir_helper.cpp:17:
/usr/include/xir/util/any.hpp:28:2: error: #error "must enable c++17"
   28 | #error "must enable c++17"
      |  ^~~~~
```
.
```
[ 28%] Building CXX object trace/CMakeFiles/trace.dir/src/util.cpp.o
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/trace/src/util.cpp: In function ‘void vitis::ai::trace::dump_to(std::vector<std::map<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> > >&, std::string)’:
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/trace/src/util.cpp:46:23: error: loop variable ‘i’ creates a copy from type ‘const std::pair<const std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> >’ [-Werror=range-loop-construct]
   46 |       for (const auto i : entry) {
      |                       ^
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/trace/src/util.cpp:46:23: note: use reference type to prevent copying
   46 |       for (const auto i : entry) {
      |                       ^
      |                       &
cc1plus: all warnings being treated as 
```
,
```
[ 38%] Building CXX object async-runner/CMakeFiles/async-runner.dir/src/batch_tensor_buffer.cpp.o
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/async-runner/src/async_runner.cpp: In destructor ‘virtual {anonymous}::AsyncRunnerImpl::~AsyncRunnerImpl()’:
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/async-runner/src/async_runner.cpp:238:23: error: ‘sleep_for’ is not a member of ‘std::this_thread’
  238 |     std::this_thread::sleep_for(std::chrono::milliseconds(1000));
      |                       ^~~~~~~~~

```
and
```
[ 99%] Building CXX object dpu-runner/CMakeFiles/dpu-runner.dir/src/imp/dpu_core.cpp.o
In file included from /home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/src/imp/dpu_session_imp.cpp:22:
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/util/include/vitis/ai/weak.hpp: In instantiation of ‘struct vitis::ai::invoke_initialize_if_possible<vart::dpu::DpuKernelDdr, void>’:
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/util/include/vitis/ai/weak.hpp:82:51:   required from ‘static std::shared_ptr<_Tp> vitis::ai::WeakStore<K, T>::create(const K&, Args&& ...) [with Args = {const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, xir::DpuController*, long unsigned int}; K = std::__cxx11::basic_string<char>; T = vart::dpu::DpuKernelDdr]’
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/src/imp/dpu_session_imp.cpp:78:75:   required from here
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/util/include/vitis/ai/weak.hpp:68:46: error: ‘this’ pointer is null [-Werror=nonnull]
   68 |     T, my_void_t<decltype(((T*)0)->initialize()),
      |                           ~~~~~~~~~~~~~~~~~~~^~
In file included from /home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/src/imp/dpu_session_imp.cpp:26:
/home/ubuntu/src/ai/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/src/imp/./dpu_kernel_ddr.hpp:48:16: note: in a call to non-static member function ‘virtual void vart::dpu::DpuKernelDdr::initialize()’
   48 |   virtual void initialize() override;
      |                ^~~~~~~~~~
```

xihuazhanghui, 2022-03-21:
Hi, alfonsosanchezbeato

This bug will be fix in version vitis-ai 2.0.

You can fix the compilation by:
1. modify vart/util/include/vitis/ai/weak.hpp:
--    T, my_void_t<decltype(((T*)0)->initialize()),
++   T, my_void_t<decltype(std::declval<T>().initialize()),

2. modify vart/async-runner/src/async_runner.cpp
++ #include <thread>

3. modify vart/cmake/VitisCommon.cmake
-  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=gnu++14 -Wall -Werror")
+  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=gnu++17 -Wall -Werror")

wilderfield, 2022-03-21:
Why is gcc11 a requirement?
The code was written and compiled against gcc9.

xihuazhanghui, 2022-03-21:
> 

Ubuntu 22.04 default gcc version is 11.2.0

xihuazhanghui, 2022-03-24:
@alfonsosanchezbeato 

/usr/include/xir/util/any.hpp:28:2: error: #error "must enable c++17"
28 | #error "must enable c++17"

-  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=gnu++14 -Wall -Werror")
+  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -Werror")

Can be fixed by modify vart/cmake/VitisCommon.cmake

xihuazhanghui, 2022-06-15:
This issue have been fixed in vitis-ai release 2.5.0

fpardoseco, 2022-06-20:
I have tried with Vitis 2.5 but I still got the same error:

"/opt/petalinux/2021.1/sysroots/cortexa72-cortexa53-xilinx-linux/usr/include/xir/util/any.hpp:28:2: error: #error "must enable c++17"

qianglin-xlnx, 2022-06-20:
Hi @fpardoseco 
Could you upgrade your cross-compiler to 2022.1 and try it again?
Refer to : https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc#step1-setup-cross-compiler

fpardoseco, 2022-06-21:
> Hi @fpardoseco Could you upgrade your cross-compiler to 2022.1 and try it again? Refer to : https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc#step1-setup-cross-compiler

Thanks that worked ! Then I got the following error:

fatal error: vart/dpu/dpu_runner_ext.hpp: No such file or directory
   23 | #include <vart/dpu/dpu_runner_ext.hpp>

I just commented the  line:

 #include dpu_runner_ext.hpp 

on filter.hpp and the compilation finished without errors.

xihuazhanghui, 2022-06-21:
Hi, @fpardoseco 

Which source code did you use?
because I didn't not found the filter.hpp or dpu_runner_ext.hpp under
https://github.com/Xilinx/Vitis-AI/tree/master/src/Vitis-AI-Runtime/VART
or
https://github.com/Xilinx/Vitis-AI/tree/master/src/Vitis-AI-Library

Thanks
Hui

xihuazhanghui, 2022-06-21:
Hi, @fpardoseco 

This tutorial case not maintain any more,  
Our  https://github.com/Xilinx/Vitis-AI/tree/master/src/Vitis-AI-Runtime/VART have some header and interface change. 

If you can fix the compiling issue and using it, I am glad to hear that.

Thanks
Hui

chkohn, 2022-07-14:
> @alfonsosanchezbeato
> 
> /usr/include/xir/util/any.hpp:28:2: error: #error "must enable c++17" 28 | #error "must enable c++17"
> 
> Can be fixed by modify vart/cmake/VitisCommon.cmake
> 
> -- set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=gnu++14 -Wall -Werror") ++ set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -Werror")

this fixed it for me but the fix doesn't seem to be in VAI 2.5 as stated.

I'm natively compiling for arm64 on Ubuntu 22.04 with gcc 11.2.0 on a Kria board.