Xilinx/Vitis-AI #718: problem with  tf2_mobilenetv1_imagenet_224_224_1.15G_2.0

fash120, 2022-03-19:
After I download the file, use matlab to export the weights in the .h5 file and fold the BN layer, the accuracy obtained by running the test set is greatly reduced, only about 43%, may I ask if the recall rate and accuracy given by your company are? There is such a big difference. At the same time, I would also like to ask how accurate the model can be.

wenjingk-xilinx, 2022-03-22:
hi @fash120 , 
the accuracy of tf2_mobilenetv1_imagenet_224_224_1.15G_2.0:
>  float model: Recall_1: 70.05%，quantized model: Recal_1: 68.95%

To get the same accuracy, I suggest you use the eval script in our repo. You can find related instructions in the readme.

fash120, 2022-03-22:
Thanks for your answer. I would like to ask, the relationship between the recall and precision of this model. In the imagenet test set, is it possible to equate precision with recall?

fash120, 2022-03-22:
@wenjingk-xilinx 
Thanks for your answer. I would like to ask, the relationship between the recall and precision of this model. In the imagenet test set, is it possible to equate precision with recall?

wenjingk-xilinx, 2022-03-23:
hi @fash120 ,
Here the recall equals precision.

fash120, 2022-03-27:
@wenjingk-xilinx thanks。

qianglin-xlnx, 2022-04-07:
Hi @fash120 
Is this issue solved?

qianglin-xlnx, 2022-04-11:
Hi @fash120 
It seems that we can close this issue. If you still have any problems, please feel free to reopen it. Thank you.

closed 2022-04-11 by qianglin-xlnx