Xilinx/Vitis-AI #719: LSTM support on zcu102 platform and vck190 platform

SamsonChau, 2022-03-22:
Hello,  I am working on a project that require the MobileNet v3 and LSTM model in a facial recognition application. Is there any method to run LSTM(XRNN) on the vck190 production board? What I found in Vitis AI User Guide (UG1414) and Vitis AI github only have the LSTM model example for VCK5000.

Currently I am able to quantize and compile the MobileNetv3 for the zcu102 trd and the vck190 default design, but still have issue with the quantization with the LSTm model. As the LSTM model takes the output of the MobilenetV3 to generate the output. And the following is my model: 
[image]

If the LSTM model cannot run in DPU / AI-engine, then is there are any approach for the running the model on the zcu102/ vck190 platform?

qianglin-xlnx, 2022-03-22:
Hi @SamsonChau 
ZCU102 and VCK190 are not in the supported list for LSTM model.

SamsonChau, 2022-03-23:
@qianglin-xlnx  thanks for reply,
Is it means the only way to  run the LSTM (RNN) model is the put it to the ARM CPU core in zcu102/ vck190 platform?

qianglin-xlnx, 2022-03-23:
@SamsonChau 
Yes.

closed 2022-04-24 by qianglin-xlnx