Xilinx/Vitis-AI #720: About the hard sigmoid in cpu

white0dew, 2022-03-22:
The moileNetV3 has op names hard-sigmoid，and the op is only supported in VCK190
but my hardware is kv260。
now i want to know if i deploy the hardsigmoid in cpu， the inference latency will increace how much？？ms？

zz002, 2022-03-23:
Tested a hard-sigmoid-fix that we support, and the calculation takes 498ms for a data size of 123904

Ace-cream, 2022-03-23:
> 

Hello, could you please send your test code to me so that I can test my data? Thanks a lot!

Ace-cream, 2022-03-23:
> Tested a hard-sigmoid-fix that we support, and the calculation takes 498ms for a data size of 123904

Hello, could you please send your test code to me so that I can test my data? Thanks a lot!

zz002, 2022-03-23:
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/cpu_task/ops/hard-sigmoid-fix/hard-sigmoid-fix.cpp#L48
The function in the link above is the main code you can use to test your data

closed 2022-04-14 by qianglin-xlnx