Xilinx/Vitis-AI #721: Could the pytorch model that compiled by Apache TVM run with KV260 and DPU?

youjunl, 2022-03-22:
Hi, all
I tried to make a tvm model that compiled from pytorch run with DPU but it was not successful. And there is key error about _"cpu-tf"_ as I running the deploy_lib_edge.so. It seems like a model framework problem. So I would ask if PyTorch models are supported on vitis-ai tvm api?

> xilinx-k26-starterkit-2021_2:~/notebooks/compile-quartznet$ python3 test_tvm.py 
> Traceback (most recent call last):
>   File "test_tvm.py", line 13, in <module>
>     lib = tvm.runtime.load_module("deploy_lib_edge.so")
>   File "/home/petalinux/tvm/python/tvm/runtime/module.py", line 552, in load_module
>     return _ffi_api.ModuleLoadFromFile(path, fmt)
>   File "tvm/_ffi/_cython/./packed_func.pxi", line 323, in tvm._ffi._cy3.core.PackedFuncBase.__call__
>   File "tvm/_ffi/_cython/./packed_func.pxi", line 257, in tvm._ffi._cy3.core.FuncCall
>   File "tvm/_ffi/_cython/./packed_func.pxi", line 246, in tvm._ffi._cy3.core.FuncCall3
>   File "tvm/_ffi/_cython/./base.pxi", line 163, in tvm._ffi._cy3.core.CALL
> KeyError: "KeyError: ('cpu-tf',)\nAt:\n  /usr/lib/python3.8/site-packages/pyxir-0.3.1-py3.8-linux-aarch64.egg/pyxir/runtime/runtime_factory.py(60): build_runtime\n  /usr/lib/python3.8/site-packages/pyxir-0.3.1-py3.8-linux-aarch64.egg/pyxir/base.py(343): build\n  /usr/lib/python3.8/site-packages/pyxir-0.3.1-py3.8-linux-aarch64.egg/pyxir/base.py(489): build_online_quant_rt_opaque_func\n  /usr/lib/python3.8/site-packages/pyxir-0.3.1-py3.8-linux-aarch64.egg/pyxir/opaque_func.py(113): opaque_func_wrapper\n  /home/petalinux/tvm/python/tvm/runtime/module.py(552): load_module\n  test_tvm.py(13): <module>"

qianglin-xlnx, 2022-03-23:
Hi @elliott-xilinx  could you please help to check this issue? Thank you.

pebush, 2022-09-25:
Hi, I also would like to ask the situation.

I made a "tvm_dpu_cpu.so" via onnx: https://tvm.apache.org/docs/how_to/compile_models/from_onnx.html

But, i enconter the same problem... 

> Traceback (most recent call last):
>   File "/home/root/workspace/tvm-vai/test/run_mxnet_resnet_18.py", line 152, in <module>
>     run(file_path, shape_dict, iterations)
>   File "/home/root/workspace/tvm-vai/test/run_mxnet_resnet_18.py", line 95, in run
>     lib = tvm.runtime.load_module(file_path)
>   File "/home/root/workspace/tvm-vai/tvm/python/tvm/runtime/module.py", line 613, in load_module
>     return _ffi_api.ModuleLoadFromFile(path, fmt)
>   File "tvm/_ffi/_cython/./packed_func.pxi", line 331, in tvm._ffi._cy3.core.PackedFuncBase.__call__
>   File "tvm/_ffi/_cython/./packed_func.pxi", line 262, in tvm._ffi._cy3.core.FuncCall
>   File "tvm/_ffi/_cython/./packed_func.pxi", line 251, in tvm._ffi._cy3.core.FuncCall3
>   File "tvm/_ffi/_cython/./base.pxi", line 181, in tvm._ffi._cy3.core.CHECK_CALL
> KeyError: "KeyError: 'cpu-tf'\nAt:\n  /home/root/workspace/venv01/lib/python3.9/site-packages/pyxir-0.3.5-py3.9-linux-aarch64.egg/pyxir/runtime/runtime_factory.py(60): build_runtime\n  /home/root/workspace/venv01/lib/python3.9/site-packages/pyxir-0.3.5-py3.9-linux-aarch64.egg/pyxir/base.py(348): build\n  /home/root/workspace/venv01/lib/python3.9/site-packages/pyxir-0.3.5-py3.9-linux-aarch64.egg/pyxir/base.py(498): build_online_quant_rt_opaque_func\n  /home/root/workspace/venv01/lib/python3.9/site-packages/pyxir-0.3.5-py3.9-linux-aarch64.egg/pyxir/opaque_func.py(113): opaque_func_wrapper\n  /home/root/workspace/tvm-vai/tvm/python/tvm/runtime/module.py(613): load_module\n  /home/root/workspace/tvm-vai/test/run_mxnet_resnet_18.py(95): run\n  /home/root/workspace/tvm-vai/test/run_mxnet_resnet_18.py(152): <module>"