Xilinx/Vitis-AI #723: How to design a DPU op, such as hard sigmoid?

Ace-cream, 2022-03-23:
My hardware is KV260, but its DPU doesn't support hard sigmoid. I want to know how to design such a DPU op, which can be supported by KV260. Thanks a lot!

qianglin-xlnx, 2022-03-23:
Hi @Ace-cream 
Yes, the DPU on KV260 doesn't support hard sigmoid op. It will be supported in VAI2.5.
So do you want do design a hardware IP for hard sigmoid? or Software OP implementation?
For software, you can refer to https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/cpu_task/ops/hard-sigmoid-fix/hard-sigmoid-fix.cpp

Ace-cream, 2022-03-23:
> Hi @Ace-cream Yes, the DPU on KV260 doesn't support hard sigmoid op. It will be supported in VAI2.5. So do you want do design a hardware IP for hard sigmoid? or Software OP implementation? For [...]

Thanks a lot@[qianglin-xlnx](https://github.com/qianglin-xlnx) for your answer. Could you please tell me when the VAI2.5 will be published?

qianglin-xlnx, 2022-03-29:
Hi @andyluo7 
Do you know when the VAI2.5 will be released?

qianglin-xlnx, 2022-04-14:
Hi @Ace-cream 
So far as I know, VAI2.5 will be released around the end of June.

closed 2022-04-14 by qianglin-xlnx